d-ronin/dRonin #532: CDC/VCP driver does not work on Windows 10

droidicus, 2016-01-25:
It looks like in Windows 10, Microsoft has changed the way that USB modem interfaces work, and the current inf based driver install will no longer work:

https://msdn.microsoft.com/en-us/library/windows/hardware/dn707976(v=vs.85).aspx

I can verify that a Sparky2 is showing up as a valid COM port in the device manager, but I cannot get VCP telemetry working on my setup. My guess is that something will need to change in the GCS, but I have not been able to find what up to this point.

tracernz, 2016-01-25:
Can also confirm the same issue on Windows 10 using the Microsoft auto-installed driver.

tracernz, 2016-01-25:
Windows 10 driver in Windows/Inf/Usbser.inf for reference: https://gist.github.com/tracernz/26a5c4343cec83462fed

tracernz, 2016-01-25:
If I open the port in Putty it's spitting out data so looks like this is an issue with GCS or QSerialPort.

tracernz, 2016-01-25:
Ahhh, it works on my compiled-from-next GCS, but not with the release one...

tracernz, 2016-01-25:
Also working on the Hotfix artifact: http://jar.lyle.org:8080/job/dronin/102/Nodes=winx86/

mlyle, 2016-01-25:
@tracernz That sounds exceedingly odd :/

tracernz, 2016-01-25:
I think I figured it out. Pretty sure that was the first time I ever plugged a board into that Windows install, and then IIRC I rebooted to installed Windows updates after testing the release but before testing next and the hotfix. @droidicus Try rebooting?

droidicus, 2016-01-27:
@tracernz no joy, still doesn't work here :-/

ragd011, 2016-02-16:
The only issue I've had was having to drop into a boot option that disables the requirement for driver signing while I install the GCS.  Then I reboot win 10 normally and everything is detected fine.

http://www.howtogeek.com/167723/how-to-disable-driver-signature-verification-on-64-bit-windows-8.1-so-that-you-can-install-unsigned-drivers/

is this what people are experiencing or am I not understanding the problem correctly?

tracernz, 2016-02-16:
You don't need to install drivers on Windows 10, they are included by default.

droidicus, 2016-02-16:
I was able to get VCP working this weekend for the Debug console on Win 10 using the default system driver (no driver installation). Will try the VCP telemetry again tomorrow.

mlyle, 2016-02-16:
@droidicus any idea what you did differently vs. when it didn't work before?

droidicus, 2016-02-16:
@mlyle I didn't do anything differently, just set the VCP port to the debug console, and fired up putty. It worked the first shot.

tracernz, 2016-02-16:
So it looks like a telemetry/configuration issue rather than a CDC driver issue?

droidicus, 2016-02-16:
Yes, I think so.

mlyle, 2016-03-21:
I had marked this documentation, but in re-reading the issue I don't see an actionable doc issue.

closed 2016-03-21 by mlyle