d-ronin/dRonin #535: Radio Setup Wizard

pug398, 2016-01-25:
A couple of things noticed in Renatus. If it has been already corrected then just ignore. On the first page of wizard spelling of "screeen" needs correcting. On the "reversing" page could the contrast of the arming switch and flight mode switch be increased to match that of the previous pages? They are hardly visible on the reversing page (cosmetic). Again if already corrected then please forgive me for not keeping up you guys are moving at a lightning pace.

mlyle, 2016-01-25:
Of course :) this is a good report-- never hesitate to open an issue.

Relates to #519

mlyle, 2016-02-11:
8e2472a126 fixed 'screeen' ; #519 covers the remaining switch issue.

closed 2016-02-11 by mlyle