d-ronin/dRonin #536: GCS: Test Motor Outputs behavior

readerror67, 2016-01-26:
One thing that I would like to see fixed, mostly just as a safety issue is the sliders in the motor output page.

If you enable the 'test motor output' option, the motors don't start spinning until you move the slider. The problem is if you slider is at the point in which the motors start spinning or slightly above, moving it to the minimum will cause the motor to spool up.

A simple solution would allow the slider to go from its current neutral to the minimum value without enabling pwm output or anywhere in between. PWM output would only be enabled if the slider transitions to the right (increasing).

mlyle, 2016-01-26:
Not marking this bug/safety because I think it deserves some discussion.  I'm a bit skeptical about changes here, but would love to hear what others think.

jhitesma, 2016-01-26:
This has always bugged me.  But I'm not sure what the proper fix would be.  Seems to me that if you enable motors and the sliders are set at a value that should make the motors spin...then they should spin.

I kind of like the way *flight does it with the settings and the output tests being two very separate operations.  That does make setting the neutral interactively a bit harder (and in *flight it's not even exposed in the GUI IIRC.)  But it seems to make the output tests safer since they always start at 0 and you have to move them up manually.

readerror67, 2016-01-26:
Also, a 'master' slider would be ace, maybe a master min/max as well.
Properly set ESCs should have identical values anyways,

dustin, 2016-01-26:
A max slider would be quite grand.  It's too easy to accidentally slip and
send the things spinning madly.

avs009, 2016-01-26:
Hi, maybe a huge alarm when change spinner, is fine. i almost hurt myself when inverted pwm, and motors spin at max speed.

tracernz, 2016-01-26:
Minimal fix seems to be just reset the sliders to min upon checking the test box?

pug398, 2016-01-26:
Esc even if done identical do not always start at the same point hence each one is adjustable. There are suppose to be no props on at this point. Although it does use the slider position as the start point each time so you have to always add the offset I have never seen one spin up by moving slider to the left. Is this case using something other than pwm out?

readerror67, 2016-01-26:
@tracernz that may work, assuming they are restored to the original state afterwards.

tracernz, 2016-01-26:
Yeah, so their function is a bit weird since they're the neutral value sliders... It's not really so simple.

readerror67, 2016-01-26:
@pug398 Moving the slider to the left does indeed start the motor. For example if the min is 125 and neutral is 133, if you move it to 130 (or anywhere in betweeen 133 and 125) the motors will spin.

pug398, 2016-01-26:
So with syncpwm? It does not do it with pwm from what I can see.

readerror67, 2016-01-26:
@tracernz good point, it seems we are currently re-purposing a settings page for testing which may not be as flexible or ideal.

Would having a motor test tab be a better option? Possible useful for calibration and basic checks without impacting/changing the current configuration of things.

tracernz, 2016-01-26:
It will occur any time if your neutral value is far enough above the point at which the motors spin because as soon as you grab the slider it will update the actuator command.

pug398, 2016-01-26:
Ok I understand now. It has always been that way and sets the output to what the slider is set for. You have to first move the slider left or right to update the output.

ekaszubski, 2016-01-27:
@tracernz 
> Minimal fix seems to be just reset the sliders to min upon checking the test box?
This seems undesirable for servos as the min value is likely a high strain and/or high displacement value. IMO any fix like this should be conditional on output type and should be avoided for servos.

jhitesma, 2016-01-27:
Yeah, resetting them upon checking the box doesn't seem like a good option since that would mean having to reset neutral each time.

Still feels to me like this should be separated out as using one UI element for two different functions is really the big problem.

A separate "diagnostics" tab with the motor test functionality really seems like the best solution to me.  Would be nice to add a gyro graph on there as well like *flight's configurator has since I find that useful for finding rough spots on motors and tracking down vibrations.  Getting off topic, but if we did move to a separate diagnostics page I'd love to see live stick inputs there as well.

mlyle, 2016-01-27:
> Still feels to me like this should be separated out as using one UI element for two different functions is really the big problem.

Yah, on the one hand that's true.  On the other hand, the most common use for this (outside of planes) is to find the neutral value-- so in that case the functions are the same.

mlyle, 2016-09-01:
IMO wontfix for now.  If other people hate this reopen.

closed 2016-09-01 by mlyle