d-ronin/dRonin #547: Some UI elements are squished with the netbook stylesheet

mngrif, 2016-01-27:
Some UI elements are squished with the netbook stylesheet. Screenshot is attached. Screen resolution is 1024x600 which is awfully tiny.

Thank you
[image]
.

mlyle, 2016-01-27:
Thanks for the report!

That one indeed looks bad.  Any other really bad examples?  This one we should be able to just put a scrollbar on the panel.

mngrif, 2016-01-27:
After a thorough review of everything I can find to review, I found two more display issues.

Thank you!

[image]
[image]

tracernz, 2016-01-28:
Should be a case of setting minimum sizes and checking for appropriate size policies on these widgets.

mlyle, 2016-02-02:
@tracernz on the last one, can you repro-- seems like it'd be a linux stylesheet thing.

tracernz, 2016-02-04:
I've seen blacked out things in the vehicle widget like that before but it's never repeatable from what I can tell.

PR #591 (2016-02-04): GCS: A couple of small UI tab changes
fixes #547 fixes #574

[[image]](https://reviewable.io/reviews/d-ronin/dronin/591)
commit c2a315ef (2016-02-04): gcs: attitude tab: adjust to fit on small screens

fixes #547
closed 2016-02-04 by tracernz