d-ronin/dRonin #564: Flight: sensors: Identify why there is AQ32-specific code

mlyle, 2016-01-29:
For mag handling / fail detection.  Why is AQ32 different?

@jihlein any thoughts?

jihlein, 2016-01-29:
From memory since I can't get to the code now:

Onboard i2c handles baro and mag. If the port init or sensor fails, that ends in a panic since the board is showing a problem. 

AQ32 has an external i2c port that's not shared with uart pins. If hooked to a mag that is not powered, I did not want the board to lock in a panic state that is not easily recoverable. So I used the i2c cwa block to indicate an issue with external i2c, and the sensor i2c block to indicate an issue with the external mag sensor. This was because the issue is likely lack of power and no need to panic the board into a locked up state. 

There may be a cleaner way to do this, it was my first change after porting the code to the AQ32 target.

mlyle, 2016-01-29:
@jihlein it'd be better if this was a mechanism shared between all external mags instead of being unique to AQ32.

jihlein, 2016-01-29:
Fully agree. I'll look into it. I know Quanton is setup this way.  Also Sparky ext mag is broken, I'll tackle that one first, something I've been meaning to do.

mlyle, 2016-01-29:
@jihlein Much appreciated!

jihlein, 2016-01-30:
@mlyle, how do you feel about I2C initialization never failing in a panic state, but setting the I2C CWA block color accordingly?  This way both internal and external I2C buses would behave the same way, and we'd loose one or two panic codes also.  Internal sensors would still end in a panic if they failed initialization, external ones would set the sensor block red.

mlyle, 2016-01-30:
I've not looked at the code in detail, but that sounds like a good approach to me.

jihlein, 2016-02-02:
Reference #557 and #589.

mlyle, 2016-02-06:
Closed in #589

closed 2016-02-06 by mlyle