d-ronin/dRonin #567: Inconsistent UUIDs in AT Reports

dustin, 2016-01-29:
@droidicus reported some of his tunes not deduplicating.  He showed the following tunes as examples:

http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgPrDkgoM
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgMfhogsM

These have UUID af078a5411290a7bcc45be05afd69e18aa9aa5ca7ba5159fc7f98d5c3e9b3cdf and 9d81b71e10f42a1ecf2eb96609718660a0e783cd2ce7fe4868ad1dbd4f45f08e respectively -- for the same board.

The hash comes from 907f15747262b0f907f4ac03f7a01677fa6bbdcc which doesn't seem like it could be doing anything wrong unless the CPU ID is coming in incorrectly from the chip (?)

dustin, 2016-01-29:
I tried a variety of zeros to see if that had something to do with it.

```
>>> hashlib.sha256(''.join(chr(x) for x in [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])).hexdigest()
'15ec7bf0b50732b49f8228e07d24365338f9e3ab994b00af08e5a3bffe55fd8b'
>>> hashlib.sha256('').hexdigest()   
'e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855'
>>> hashlib.sha256('\x00').hexdigest()
'6e340b9cffb37a989ca544e6bb780a2c78901d3fb33738768511a30617afa01d'
>>> hashlib.sha256('0').hexdigest()
'5feceb66ffc86f38d952786c6d696c79c2dbc239dd4e91b46729d73a27fb57e9'
```

dustin, 2016-01-29:
This timeline is very close to when we got rid of raw chip IDs in favor of UUIDs, such that it kind of spans the conversion.  I went digging a bit and found my sparky2 which has tunes covering this time period while still lining up OK, so I don't think it's that.

http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgLvjjQkM

dustin, 2016-02-06:
This is happening to all of mine today.  So that's pretty bad.  I've updated a sparky2 and a naze32 and both showed up as new boards in usage and got disconnected tunes.

Sparky2:
[newer](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgLOozgoM)
[older](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgLvjjQkM)

Naze32:
[newer](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgMG57gsM)
[older](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgILMuAgM)

dustin, 2016-02-12:
I did an upgrade on Brain which did this as well.

New UUID:  9af89b69fb585bb43a10cbbd6249d2efb514cfcfce28fc2bc14551b15da031d0
[Example](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgKK-tAsM)

Previous UUID:  21d059c3b750e760c2c74465befd013b565e1f25df82edde73667d8354beb819
[Example](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgLnH1QgM)

The serial exported in UAVO has consistently been 4e004f000c51343231383537 since October (several snapshots out).

hashlib.sha256(binascii.a2b_hex('4e004f000c51343231383537')).hexdigest()
'9af89b69fb585bb43a10cbbd6249d2efb514cfcfce28fc2bc14551b15da031d0'

This suggests that it was perhaps incorrect for a bit and now it's fixed?  :/

dustin, 2016-02-12:
Note (because @mlyle asked):  Both of those UUIDs in the previous comment show up in the board export.  The GCS consistently reported that UUID in both AT and Usage.

I have confirmed that the same UUID is reported in the serial (4e... in example above) in each of the following revisions of my uav export:

```
2016-02-11 21:12:56 -0800 Prepping today's next
2016-01-17 21:56:05 -0800 dRonin on the frogger 18
2015-11-13 19:21:24 -0800 Adjusted battery thresholds to more useful values.
2015-10-27 21:53:52 -0700 Actually apply OSD changes (+deadband)
2015-10-27 20:51:40 -0700 OSD screen rearrangement
2015-10-26 23:50:53 -0700 Increased yaw rate to 420.
2015-10-20 20:15:24 -0700 Not going to do frsky hub
2015-10-20 19:27:30 -0700 Added frsky telemetry
2015-10-20 19:20:51 -0700 Added frsky rssi
2015-10-20 19:20:24 -0700 New tune with comp_pwm
2015-10-17 23:33:55 -0700 Move throttle idle a bit to avoid yaw input when arm/disarm
2015-10-16 11:53:47 -0700 Changed default flight mode to AcroPlus
2015-10-16 11:50:55 -0700 OSD cleanup
2015-10-16 11:05:51 -0700 Misc updates that aren't meaningful to me
2015-10-16 10:38:37 -0700 Adjust battery-based time thresholds
2015-10-16 10:37:35 -0700 Fix throttle curve
2015-10-16 09:43:20 -0700 Added MRSS 180 config
```

mlyle, 2016-03-21:
Sucks but not the end of the world.  Deferring.

mlyle, 2016-09-01:
Do we still see this?  Various telem bugs we fixed could have done this..

dustin, 2016-09-01:
I don't know how to know.  The only real way to tell is "I autotuned a
board I've autotuned before and the previous tunes aren't linked."  As far
as I know, the right thing has been happening, though.

mlyle, 2016-09-01:
Yah, that seems to be the same for me.  I have not seen it, I have not heard anyone else mention that they've not been linked.  It was very rare in the beginning and seemed to disappear.

closed 2016-09-01 by mlyle