d-ronin/dRonin #574: Basic and Advanced Stabilization order inconsistant

ufoDziner, 2016-01-30:
The order of the large boxes that contain the sliders are in a different order on the Basic tab and Advanced tabs.

From top to bottom on Basic:
Rate Stabilization
Attitude Stabilization
Stick Scaling

From top to bottom on Advanced:
Rate Stabilization
Stick Scaling
Attitude Stabilization

Stick Scaling on the Basic page should be after Rate Stabilization.

mlyle, 2016-02-02:
So I looked at this in a bit of detail.  I know it's not quite the same as the basic pane, but the intention here was to put the inner loop's stick scaling next to the inner loop, and the outer loop's stick scaling next to the outer loop.  To date, on the 'basic' pane the attitude limits haven't shown.

I'm tempted to just remove the stick scaling from the first page to make this consistent, or to ignore this. :/ what do you think?

tracernz, 2016-02-02:
Just move it up imo. The reason it's there is because it's the first thing multiwii pilots look for, which happens to be the same group that like scaled gains.

PR #591 (2016-02-04): GCS: A couple of small UI tab changes
fixes #547 fixes #574

[[image]](https://reviewable.io/reviews/d-ronin/dronin/591)
commit b2aa4569 (2016-02-04): gcs: stabilization: move rates up on basic tab

fixes #574
closed 2016-02-04 by tracernz