d-ronin/dRonin #582: Firmware flashing fails for CC3D

ekaszubski, 2016-01-31:
System
- Ubuntu 15.10 x64
- CC3D HW Rev 2 BL Ver 4

Steps to reproduce
- Check out any rev after #540 was merged (fd93c056)
- Make gcs fw_coptercontrol
- Rescue, connect, select fw_coptercontrol.tlfw, flash
- Flashing immediately fails with "Firmware upload failed"

Additional info
- Haven't tested any other FCs
- Have tested multiple revisions prior to fd93c056 back to Release-20160120 and everything works as expected

tracernz, 2016-01-31:
Can you try compiling with this line enabled: https://github.com/d-ronin/dRonin/pull/540/files#diff-f698c290bac8f66eed54b6ec8ec61803R32

ekaszubski, 2016-01-31:
@tracernz Compiling and trying now, had to fix this line https://github.com/d-ronin/dRonin/pull/540/files#diff-f698c290bac8f66eed54b6ec8ec61803R155 after enabling debug

ekaszubski, 2016-01-31:
@tracernz Can confirm the firmware flashing works with line 32 enabled and line 155 changed to:
`TL_DFU_QXTLOG_DEBUG(QString("DOWNLOAD PARTITION- SETTING DOWNLOAD CONFIG PARTITION=%0 SIZE: %1").arg(partition).arg(size));`

mlyle, 2016-01-31:
Since clicking "flash" again makes this work, removing blocker tag (easy interim workaround).

tracernz, 2016-02-01:
@ekaszubski That's concerning. Seems like it's a very subtle timing/race condition type thing if disabling the debug output triggers it...

ekaszubski, 2016-02-01:
@tracernz Yep, @mlyle had similar concerns. Looks like the workaround is really simple for now, though I never thought to try clicking flash twice until it was suggested to me (so may be a big issue for new users of GCS).

PR #524 (2016-03-04): gcs: take less of forever to connect.
Also update the connection status object more frequently so we're less
likely to spuriously lose connection from little blips / lost messages.

fixes #582 

[[image]](https://reviewable.io/reviews/d-ronin/dronin/524)
closed 2016-03-04 by tracernz