d-ronin/dRonin #584: Windows GCS build failure with -j

droidicus, 2016-01-31:
I have been experiencing GCS build failures on windows recently after using 'make all_clean && make all -j', it will work fine with -j4, and in the case of my i7 processor I believe that -j is using 8 threads. Here is an example of the end of the build process:
```
g++ -shared -Wl,-subsystem,console -mthreads -Wl,--out-implib,C:/tau/dRoninFork/build/ground/gcs/lib/dr/plugins/TauLabs/libUAVObjectUtild.a -o ../../../lib/dr/plugins/TauLabs/UAVObjectUtild.dll debug/uavobjectutilmanager.o debug/uavobjectutilplugin.o debug/devicedescriptorstruct.o debug/moc_uavobjectutilmanager.o debug/moc_uavobjectutilplugin.o  -LC:/tau/dRoninFork/build/ground/gcs/lib/dr -LC:/tau/dRoninFork/build/ground/gcs/lib/dr/plugins/TauLabs -lAggregationd -lExtensionSystemd -lUtilsd -lCored -lUAVObjectsd -LC:/Qt/Qt5.5.0/5.5/mingw492_32/lib -lQt5Widgetsd -lQt5Guid -lQt5Testd -lQt5Cored
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/build/ground/gcs/gcsversioninfo.h\" -DGCS_DATA_BASENAME=\"share\" -DQT_NO_CAST_TO_ASCII -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"C:/tau/dRoninFork/build/ground/gcs/src/plugins/uavobjectbrowser"' -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjectbrowser -I. -IC:/tau/dRoninFork/shared/api -IC:/tau/dRoninFork/ground/gcs/src/libs -IC:/tau/dRoninFork/ground/gcs/src/plugins -IC:/tau/dRoninFork/build/uavobject-synthetics/gcs -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjects -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtWidgets -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtGui -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtANGLE -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtTest -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtCore -Idebug -I. -IC:/Qt/Qt5.5.0/5.5/mingw492_32/mkspecs/win32-g++  -o debug/moc_fieldtreeitem.o debug/moc_fieldtreeitem.cpp
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/build/ground/gcs/gcsversioninfo.h\" -DGCS_DATA_BASENAME=\"share\" -DQT_NO_CAST_TO_ASCII -DQWT_DLL -DQT_PLUGIN -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"C:/tau/dRoninFork/build/ground/gcs/src/plugins/gpsdisplay"' -IC:/tau/dRoninFork/ground/gcs/src/plugins/gpsdisplay -I. -IC:/tau/dRoninFork/shared/api -IC:/tau/dRoninFork/ground/gcs/src/libs -IC:/tau/dRoninFork/ground/gcs/src/plugins -IC:/tau/dRoninFork/build/uavobject-synthetics/gcs -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjects -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSvg -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtWidgets -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtGui -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtANGLE -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSerialPort -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtTest -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtCore -Idebug -I. -IC:/Qt/Qt5.5.0/5.5/mingw492_32/mkspecs/win32-g++  -o debug/moc_gpsparser.o debug/moc_gpsparser.cpp
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/build/ground/gcs/gcsversioninfo.h\" -DGCS_DATA_BASENAME=\"share\" -DQT_NO_CAST_TO_ASCII -DQT_PLUGIN -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"C:/tau/dRoninFork/build/ground/gcs/src/plugins/dial"' -IC:/tau/dRoninFork/ground/gcs/src/plugins/dial -I. -IC:/tau/dRoninFork/shared/api -IC:/tau/dRoninFork/ground/gcs/src/libs -IC:/tau/dRoninFork/ground/gcs/src/plugins -IC:/tau/dRoninFork/build/uavobject-synthetics/gcs -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjects -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSvg -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtWidgets -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtGui -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtANGLE -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtTest -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtCore -Idebug -I. -IC:/Qt/Qt5.5.0/5.5/mingw492_32/mkspecs/win32-g++  -o debug/moc_dialgadget.o debug/moc_dialgadget.cpp
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/build/ground/gcs/gcsversioninfo.h\" -DGCS_DATA_BASENAME=\"share\" -DQT_NO_CAST_TO_ASCII -DQWT_DLL -DQT_PLUGIN -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"C:/tau/dRoninFork/build/ground/gcs/src/plugins/gpsdisplay"' -IC:/tau/dRoninFork/ground/gcs/src/plugins/gpsdisplay -I. -IC:/tau/dRoninFork/shared/api -IC:/tau/dRoninFork/ground/gcs/src/libs -IC:/tau/dRoninFork/ground/gcs/src/plugins -IC:/tau/dRoninFork/build/uavobject-synthetics/gcs -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjects -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSvg -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtWidgets -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtGui -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtANGLE -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSerialPort -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtTest -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtCore -Idebug -I. -IC:/Qt/Qt5.5.0/5.5/mingw492_32/mkspecs/win32-g++  -o debug/moc_nmeaparser.o debug/moc_nmeaparser.cpp
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/build/ground/gcs/gcsversioninfo.h\" -DGCS_DATA_BASENAME=\"share\" -DQT_NO_CAST_TO_ASCII -DQT_PLUGIN -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"C:/tau/dRoninFork/build/ground/gcs/src/plugins/dial"' -IC:/tau/dRoninFork/ground/gcs/src/plugins/dial -I. -IC:/tau/dRoninFork/shared/api -IC:/tau/dRoninFork/ground/gcs/src/libs -IC:/tau/dRoninFork/ground/gcs/src/plugins -IC:/tau/dRoninFork/build/uavobject-synthetics/gcs -IC:/tau/dRoninFork/ground/gcs/src/plugins/uavobjects -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtSvg -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtWidgets -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtGui -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtANGLE -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtTest -IC:/Qt/Qt5.5.0/5.5/mingw492_32/include/QtCore -Idebug -I. -IC:/Qt/Qt5.5.0/5.5/mingw492_32/mkspecs/win32-g++  -o debug/moc_dialgadgetoptionspage.o debug/moc_dialgadgetoptionspage.cpp
C:/Qt/Qt5.5.0/5.5/mingw492_32/bin/ccache g++ -c -pipe -fno-keep-inline-dllexport -mno-ms-bitfields -g -std=c++0x -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DBRANDING_PATH=\"C:/tau/dRoninFork/ground/gcs/../../branding\" -DGCS_PROJECT_BRANDING=\"dr\" -DGCS_PROJECT_BRANDING_PRETTY=\"dRonin\" -DGCS_PROJECT_BRANDING_HELP=\"http://dronin.org\" -DGCS_LIBRARY_BASENAME=\"lib\" -DWITH_TESTS -DGCS_VERSION_INFO_FILE=\"C:/tau/dRoninFork/buil [... cut ...] -lCored -lGLC_libd -lUAVObjectsd -lopengl32 -LC:/Qt/Qt5.5.0/5.5/mingw492_32/lib -lQt5OpenGLd -lQt5Widgetsd -lQt5Guid -lQt5Testd -lQt5Cored
Makefile:84: recipe for target 'sub-plugins-make_first-ordered' failed
mingw32-make[2]: *** [sub-plugins-make_first-ordered] Error 2
makefile:40: recipe for target 'sub-src-make_first-ordered' failed
mingw32-make[1]: *** [sub-src-make_first-ordered] Error 2
makefile:267: recipe for target 'gcs' failed
mingw32-make: *** [gcs] Error 2
```

Race condition maybe?

mlyle, 2016-01-31:
This means you ctrl-c'd a previous build while it was compiling / linking  `C:/tau/dRoninFork/build/ground/gcs/lib/dr/plugins/TauLabs/UAVObjectUtild.dll` Make on windows isn't always able to force the removal of that file.

mlyle, 2016-01-31:
(If I erroneously closed this, --- e.g. you're sure you did a full clean before like stated-- i'm sorry.  But every time I've seen this it's been because I've aborted a prior build).

droidicus, 2016-01-31:
@mlyle I have had that same problem with ctrl-c in the past as well, but I am 99% sure it is not that this time. Just to be sure I re-ran the tests, and observed the following, in the order of execution:
make all_clean && make all -j     -->     Failure
make all_clean && make all -j4   -->     Success
make all_clean && make all -j     -->     Failure

This time I saw the problem with GCSControld.dll. I also noted that it didn't fail with '-j' on one attempt, seems random. Please re-open the issue, and let me know if there is further debugging you would like me to do.

mlyle, 2016-01-31:
I guess there's a build dep missing somewhere

tracernz, 2016-02-01:
Also, j by itself is bad news, make will launch as many processes as it possibly can.

droidicus, 2016-02-01:
Not disagreeing, I was playing with build options trying to find the fastest build time possible, and was surprised when it failed. It is easy to work around, so I wouldn't put it at high priority, but I do think that it indicates a missing build dep.

tracernz, 2016-02-01:
Yeah, if the deps are correct it should still build, just slowly if you have too many threads. :P

mlyle, 2016-02-02:
I thought about this -- maybe not.  If resources run out and a process doesn't start or is abnormally terminated and make doesn't notice, you might get something like this...

droidicus, 2016-02-06:
After talking with @mlyle in IRC, I ran a test with 'make all_clean && make all -j9' and repeated it 10 times. I was not able to reproduce the issue while doing this, and so am closing this with the instruction of "friends don't let friends use -j".

closed 2016-02-06 by droidicus