d-ronin/dRonin #587: Telemetry over radio Safety Issue

ukmook, 2016-02-01:
I use oplinkmini with a bluetooth module to allow my surface tablet to communicate with sparky 2 over 433mhz set at 25mw using bind wizard. After connecting the tablet and arming the front of the quad came up when applying throttle but the rear stayed down, motors stayed at idle, when checking in gcs the compass was spinning constantly at a high rate and also the artificial horizon moving in all directions. The Control Radio had lost all settings and the motors were at 1000 - 2000 were they should have been on one-shot. Disconnected everything, plugged usb in and all looked at it should be in GCS, Took off and flew an Autotune. 
Checked at home with radio again and would not regconise board but some of the settings were readable. Loaded TauLabs and everything works as it should using the above method. Flew an autotune and gcs telemetry working fine.
I use the GCS at the field to make changes to autonmous flight modes.

mlyle, 2016-02-01:
@pug398 have you ever seen anything like this / have any idea what it could be?  You've been flying with telemetry on Sparky2 quite a bit, right?

I have not done any telemetry flights lately but have quite a bit of time on the bench.

pug398, 2016-02-01:
I have had no issues with Sparky2 telemetry only over an oplink with bt. What air rate did you use in the RFmBind wizard (100k, 192k)? I have seen win10 usb conflict cause issue with gcs but none with OPLM+bt and GCS.

ukmook, 2016-02-01:
I used 100k at 25mw 0-250. Have tried everything i can to make it work correctly. using an hc-06 bluetooth module at 57600, connected to main port on oplink, flexport powered by 5 volts, pigtail to 433mhz 3dr receiver. i can post video of setup working on taulabs master release if that would be of any help.

mlyle, 2016-02-01:
I can't reproduce here on dRonin :/ just briefly hovered sparky2 with telemetry up

pug398, 2016-02-01:
This was just a power up and go test so bumped it to the top. Using win10 tablet with builtin bt connected to a oplm+bt on sparky2 (renatus3) 57k. Ran a pack down and never lost connection or had any issue. Switched in and out of different flight modes, did some PH and it worked ok. 100k, 100mw, telem only, 0, 250, sparky2 coordinator. I have run this setup for some time and have not had any major issues with it. PPM+Telem is a different story. @ukmook you don't have a bt keyboard/mouse usb dongle plugged in or something else like that? That is the only thing that I have seen drive the dRonin gcs crazy.

ukmook, 2016-02-01:
Think after talking to mlyle we might have solved it by erasing the settings partition. Will Test and close if all fixed tomorrow. Thanks for looking into this pug.

mlyle, 2016-02-01:
@ukmook I know you were hesitant to open an issue, but thanks for doing so.  It helps visibility, and even though we're likely to close out this issue it still helps us know about the kinds of things people can run into so we can improve the software.

ukmook, 2016-02-01:
As requested adding uav xml and bin files, still having problems when connected over radio

[issue 587.zip](https://github.com/d-ronin/dRonin/files/113167/issue.587.zip)

mlyle, 2016-02-01:
@ukmook I will continue to skim this and play.. Do you have any idea what you did immediately before it broke?

ukmook, 2016-02-01:
I did six point calibration over radio, i have done this before with TL with no issues. then i rebooted the quad and the issues started occurring again.

ukmook, 2016-02-02:
Have redone the whole setup using USB as requested, the after reboot the config screen comes up with unrecognised board. have save a log file to the zip as well as Screen captures taken after the issue arises.
[TauLabs-2016-02-02_22-13-40.zip](https://github.com/d-ronin/dRonin/files/114767/TauLabs-2016-02-02_22-13-40.zip)

mlyle, 2016-02-02:
Identified some problems with bluetooth on pipxtreme/oplm that are being opened in another issue.

I believe the difficulties experienced here with flight performance relate to some of these pipxtreme bugs when used over bluetooth (saving objects back that were not successfully retrieved).

closed 2016-02-02 by mlyle