d-ronin/dRonin #588: Sparky 2 receiver not working

Fvk1960, 2016-02-02:
I use a walkere rx1002 with ppm encoder board as used for apm.
Sometimes the ppm channels RC INPUT are detected but most times not.
Checked wiring several times.
Setup is sparky2 , ppm. gps on mainport. external mag on flexport.battery monitor.

Fvk1960, 2016-02-02:
Oh I am also using oneshot125 esc

mlyle, 2016-02-02:
@Fvk1960 Hmm.  PPM is fairly reliable from what I've seen.

Do you have a battery connected?  S2 doesn't power the radio from USB power.

ekaszubski, 2016-02-02:
I have a rx1002 on hand if we need extra testing there, but I do *not* have a PPM encoder board.

ekaszubski, 2016-02-02:
@mlyle If using the dedicated receiver port set to 3.3V, sparky2 will actually send out power to the receiver when only connected via USB (I have a spektrum satellite receiver wired up this way). Not sure what happens if you have the receiver port bridged the other way (VCC_UNREG).

mlyle, 2016-02-02:
VCC_UNREG and USB power are combined via diode OR.  RX1002 wants 4.8V or more.

Fvk1960, 2016-02-02:
On the board there is a jumper to select the powersource either 3.3v or 5v unreg from output connectors. Mine 5v is bridged to the dedicated receiver port (only port available no PCM inputs on sparky2). So receiver is powered from esc motor ports.
However the jumper was a 10k smd R15 while on document it says do not populatie R15/R17 because it is jumper pads. So i removed R15 to het receiver working . Question is the receiver port input on the board 5v tolerant?

tracernz, 2016-02-02:
> Question is the receiver port input on the board 5v tolerant?

Yes.

closed 2016-02-02 by mlyle

Fvk1960, 2016-02-02:
Like mentioned in the naze with cleanflight when connecting sonar you need to add 1k resistor.
So I also now added a 1K resistor to the ppm signal wire.
It now works just fine.
Is this reveiling information  ?

mlyle, 2016-02-02:
@fvk1960 if you had a 10k resistor across the jumper pad that powers the receiver, that is what was wrong.  10k impedance will not satisfactorily power anything.  It is meant to be a small blob of solder on that jumper.

tracernz, 2016-02-03:
Should be 0 ohm resistor/jumper or a blob of solder.