d-ronin/dRonin #597: Calibration over radio: can't save afterwards

mlyle, 2016-02-02:
Tried current next with pipxtreme + sparky2, 100kbps, 100mW ; mag calibration proceeded fine over radio.  Subsequent attempts to "save" did not (could not save any object, even from other panes).  I wonder if some objects are left "turned up" in rate.  After turning attitude down from 100ms to 333ms, was able to save objects again.

On a fresh session not after 6 point cal, was able to save.  Then after 6 point cal could not.

Taulabs/taulabs#1474 and TauLabs/taulabs#1475 are possibly related.

jpbarraca, 2016-02-04:
Also happens when configuring through a Bluetooth radio.

mlyle, 2016-03-05:
This is still a little dodgy but works.. deferring to look at it more later, though.

PR #754 (2016-03-06): gcs: make calibrations work better over slow radio links
fixes #597 

It worked before at 100kbps but not 64kbps with the telemetry updates (didn't used to, but recent telemetry changes fixed it).

With these changes, it works nicely over 64kbps.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/754)
closed 2016-03-06 by dustin