d-ronin/dRonin #604: Motors continue to run after "Motor Test" unchecked

SystemsGuy, 2016-02-03:
On Release-20160120: 20160119-2831e945 GCS, with a Brain, setup for x-quad, oneshot, PPM input, when using the "Test Outputs"" select box on the Output Tab, I am able to use the sliders to test the motors.  If, however, I "Uncheck" the Test Outputs", the motors continue to run.

mlyle, 2016-02-03:
I have not seen this, --- need to look into it.

mlyle, 2016-02-03:
Can you please get a screenshot when this is happening?

One thing I thought of--- depends on motor calibration, but if the min value is high enough for motors to run, and the neutral is used to "get them started"... even after disabling output test they'll keep spinnin'

mlyle, 2016-02-23:
We talked about this on IRC and found that it was a bad min value that caused it.

closed 2016-02-23 by mlyle