d-ronin/dRonin #606: Change "futaba sbus" to sbus in setup wizard

mlyle, 2016-02-03:
Now 3 different people on IRC have tripped over this.

tracernz, 2016-02-03:
But it is Futaba even if some cloners reverse engineered it :/. Probably worth the misnaming for the usability win though I guess.

mlyle, 2016-02-03:
Yup.  It's also a good question as to whether we can legitimately use their brand names and quasi-logos and not be legally exposed.

jpbarraca, 2016-02-04:
The same applies to Graupner and Spektrum. Also, if we remove the reference to the brand, sometimes people will struggle to associate the protocol with what they have. This can probably be solved by creating a wiki page.

henn1001, 2016-02-04:
Just for reference - Librepilot does it like that:
[image]

tgreer, 2016-02-23:
@henn1001 that works quite nicely. Though we'd need a list of types<>makers... s.bus for like frsky, etc.

PR #686 (2016-02-24): GCS:SetupWizard: Rename receiver types
Makes them a little more general/manufacturer agnostic. Fixes #606.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/686)
closed 2016-02-24 by mlyle