d-ronin/dRonin #610: Vibration test sucks.

mlyle, 2016-02-04:
* The way the configuration works in GCS is extremely fidgety; requires restart of GCS to pick up a new number of spectrogram bins / easy to provoke to crash.
* Only can seem to run it with 64 or fewer bins.  With more bins, data never shows up.
* One UAVO instance per output point.
* With that, what's one left with?  64 bins is not much spectral data. 
* Why even do the FFT flight side to begin with?  There's no buffering / accumulation of the spectral energy after FFT.  There's no magical downsampling.  It boxcar's accelerometer data, and then when it has 64 points, dumps 64 points into the FFT and gets bins out.  Might as well just hand the raw accumulated accel data to GCS to do less retarded things with it.

I nominate for removal.

jpbarraca, 2016-02-04:
From a user perspective:

- It's fidgety but works and is rather useful.
- With a Sparky 2.0 I'm also experiencing loss of data with more than 64 bins.

Do we have the bandwidth in the telemetry channel to report vibration data more often (1-5ms)?

jpbarraca, 2016-03-22:
@mlyle Please check PR #822 as an initial attempt at this issue.

PR #893 (2016-04-18): New vibrationAnalysis module and Spectrogram with FFT
New take at #610 

VibrationAnalysis Module
- module will only acquire and send data (no FFT)
- using 16 bits to reduce memory footprint. Floats scaled without clipping (considering +/-16g accel)
- module will reallocate buffers according to the number of samples required or allocates a single buffer (using an ifdef)
- reconfiguration will only take place at the start of an acquisition period so that jitter is minimised
- supports up to 1024 bins and can potentially support much more. Be careful of the bandwidth required.

GCS Spectrogram
- [... cut]
closed 2016-04-18 by mlyle