d-ronin/dRonin #621: Quantize dT based on sensor rate

droidicus, 2016-02-07:
The dT used system wide for autotune, stabilization, attitude, etc should be centralized, and measurements quantized to the sensor rate. This will help to remove the jitter inherent in the RTOS timer currently used for these purposes. Will also lay the framework for some more advanced algorithms such as accumulator based autotune, and more.

Superset of #113

mluessi, 2016-02-07:
Instead of quantizing it, I wonder if it would be better / cleaner if the sensor drivers published the sampling rate, so it can be used elsewhere.

droidicus, 2016-02-07:
@mluessi I agree, I was actually surprised that the sensor drivers didn't make this available.

tracernz, 2016-02-08:
I will put this in with new MPU driver: https://github.com/tracernz/dronin/commit/1a513a8e60374c485891541801ba6e9cd467f383

mlyle, 2016-02-08:
I think if we're doing this, we need to measure and make sure we're actually getting samples at that rate.

tracernz, 2016-02-08:
We don't want to use real time data in our calculations though, it will just introduce additional noise. Just raise an alarm if the samples are consistently more than a certain percentage over nominal maybe?

mlyle, 2016-02-08:
Yes, we definitely don't want to use it most places (-- well, we want to use it quantized to the interval in many places so we handle droppage better --) .. But we do want to notice if we think we'll be getting 333 samples per second and get 250.  Or 500.  And refuse to arm, etc.

mlyle, 2017-01-13:
In large part this is done :D

closed 2017-01-13 by mlyle