d-ronin/dRonin #635: Flight: Merge Colibri and Quanton targets

VArcht, 2016-02-09:
Talking with icee he mentioned dropping colibri as a target and that colibri and quanton were pretty much the same and that at one point it was possible to load quanton fw on the colibri.

tgreer, 2016-02-23:
@VArcht how different are they? Is this just a case of relabelling things, or is there some subtle differences we need to account for?

ernieift, 2016-03-12:
The hardware is very similar. But the bootloader version differs here. So there is fewer space for firmware on the colibri. As a workaround you can flash the colibri to a quanton.

mlyle, 2016-03-12:
We're working on auto-bootloader-update as part of the upgrade process, so this will get easier.

mlyle, 2016-05-28:
Colibri is gone gone gone.

closed 2016-05-28 by mlyle