d-ronin/dRonin #642: power-up issue with a sparky2

thunderdan, 2016-02-10:
After setup for my fixed wing craft I observe a problem when I connect my flight battery without connecting USB.  I have no issue when I connect USB first, followed by battery second.  There is a problem if I just connect the battery only.

The problem looks like a glitching of the power rails.  The power LEDs blink abnormally.  The servos glitch and then end up at their extreme positions instead of centered like they should be.  

The battery is attached to an ESC with BEC and the cable then attached to servo output channel 3.

I'm using the dronin release from January 26th 2016 for both the gcs and the firmware onboard the plane.

Things I've ruled out:
  low battery (i've used a fresh 3 cell battery charged to 12.6V)
  incorrect servo connnections (control of surfaces and throttle is correct w/ usb plugged in)

Any suggestions on things to try?

thunderdan, 2016-02-10:
I added a video of the problem on youtube.  Link is http://youtu.be/LPm7VbA4zF0

thunderdan, 2016-02-10:
Update:

I erased all settings from board, ran through setup again, and tried again and it worked as expected this time.  Maybe the default initialization of the board has the PWM outputs set to 1000 us and once the firmware/settings load then the netrual values are used (1500 us for my plane). 

I'm not sure anyone should start to work on this unless I can reproduce the problem again.

thunderdan, 2016-02-14:
I have seen this problem again when I took my plane out to fly today.  I believe it is a power droop issue.  I have two batteries that I have used for about a year.  They are the zippy compact 3 cell, 2200mah, and 25C... and yes, they are charged to capacity (12.6V).  With these batteries, the sparky2 does not complete power-up and goes into a loop (see youtube video).

When I tried a new battery in my collection, venom 3 cell, 1300mah, 20C, there are no problems.  Power-up occurs just fine multiple times in a row.

Here are the things drawing power on my board:
* 3 servos on the pwm output pins
* 1 receiver (frksky d4r-ii) on the receiver port
* GPS (neo 8m based) on the mainport

If initialization took into account planes with servos and didn't drive them to 1000 us before setting them to neutral at 1500 us, it might fix this issue?  I know on quad copters the motors all need to be off (1000 us) but on planes output needs to be set to a midway point neutral (1500 us).

Last but not least, thanks for the great project.

thunderdan, 2016-02-15:
I confirmed the same problem exists on fw/sw from Tau Labs 2015-11-23 release.

mlyle, 2016-02-15:
@thunderdan -- we do default to 1000us pulses, but we don't ever send 1000us during initialization.  We read config, (or reset to defaults which is min=1000us) ... then send it.

But one thing I've noted is that Sparky2 has some kind of reset glitch thing in its startup code, which will generate a runt pulse.  How long are things driven to one rail?  Does it get better on its own?

@tracernz -- wondering if it's this issue, https://github.com/TauLabs/TauLabs/issues/1821 --- but it always was just a little glitch for me, not a major deflection / anything sustained?

thunderdan, 2016-02-16:
> @tracernz -- wondering if it's this issue, TauLabs/TauLabs#1821 --- but it always was just a little glitch for me, not a major deflection / anything sustained?

For my set up, it can get stuck in a loop with the servos fully deflected.  When the power up completes correctly it looks like a glitch to me also.  I added a YouTube video to show the problem (better than words).

I not sure of the quality of my hobby king ESC so that could be a contributing factor.  

-- Dan Morgan --

mlyle, 2016-02-16:
OK.  We'll use this issue to try and work on Sparky2 reset behavior.  I think the bootloader tries to drive the pins low on sparky where on most other targets we leave it floating high.  

(I am curious--- not worth a test / over-deflecting control surfaces... but do you know if they are just 'left' fully deflected or if they are still being actively driven to full deflection / buzzing when stuck?  Because I could see us generating something that looks like a single full-deflection pulse during the reset sequence and then if reset fails to complete because of power rail issues leaving the motors stuck).

In any event, it seems like you need a bigger BEC and/or more capacitance on the servo power rail/flight board to avoid power quality issues from causing controller resets. We can make the behavior better, but flight board resets are bad and there's a limit to how sane it can be.

thunderdan, 2016-02-17:
Thanks for the comments Michael.

It happens fast but it seems to drive them to fold the flexion and then the reset occurs.

 I took slow motion video in case that helps to characterize it further 

http://youtu.be/7_23MPEvunk
http://youtu.be/3-pQLFIrb3k

-- Dan Morgan --

mluessi, 2016-02-17:
@thunderdan how are you powering the sparky2 and the servos? Like @mlyle suggested you may need a bigger BEC; the servos may draw too much power, which causes a brownout reset of the flight controller due to the voltage dropping.

thunderdan, 2016-02-18:
@mluessi BEC is built into the ESC that came with a floater jet I bought (and since crashed).  I agree that a BEC with more capacitance on the 5V out line would be one way fix this.

I did an experiment and now I think I know what is going on.  The pins are driven to this state during the bootloader portion of boot before the fw loads.  I loaded GCS, ran "rescue", plugged in sparky2 over USB, then plugged in the battery on my plane.  The servos were drive to full deflection and stayed put.

I was thinking I should try to build a custom boot loader.  What do you folks think... am I on the right track?

mlyle, 2016-02-18:
@thunderdan We need to study what the bootloader does.  It does not attempt to generate PWM pulses.

What is probably happening is this:  during reset, the STM32 lets the pins float high.  As soon as we're running, we drive them low.  This short blip probably looks like a pulse to the servos.

The bootloader maybe should leave the pins alone (I think other bootloaders do this), or maybe should drive them high instead.  OTOH, this is dangerous in other situations.  E.g., on brushed quads, 'high' means "let the motors spin". 

This requires careful thought.  @tracernz and I have already put some effort in on this.  The obvious "dumb" thing to do on sparky2-- be the same as other bootloaders-- is probably the right thing but may not be unconditionally safe because of the existence of brushedsparky.  (Alas, I don't think @peabody124 elected to use the same range of pins for Sparky2's FET outputs and brushedsparky's FET outputs).

thunderdan, 2016-02-19:
Update:
I found code that indicates the bootloader for sparky2 drives PWM outputs low... based on the comment at least.

flight/targets/sparky2/bl/pios_board.c

`        // Make sure all the PWM outputs are low
        const struct pios_tim_channel * channels = pios_servo_cfg.channels;
        uint8_t num_channels = pios_servo_cfg.num_channels;
        for (int i = 0; i < num_channels; i++) {
                GPIO_Init(channels[i].pin.gpio, (GPIO_InitTypeDef*) &channels[i].pin.init);
        }
`

tracernz, 2016-02-19:
No matter what you do the STM32 goes high-Z with weak pullup enabled during reset, it's probably best to continue this until we initialise PWM in the firmware. The brushed sparky should have stronger pulldown resistors on the gates, and indeed it does so it's a non-issue IMO.

pug398, 2016-02-19:
The servo controller will only activate on valid pwm input AFAIK. Are we sure during reset it is not sending default output values? With the default output set to off (0) instead of 1000-2000 there appears to be a different behavior but I only tested with motor setting not servo.

tracernz, 2016-02-19:
Original issue for reference: TauLabs/TauLabs#1821

pug398, 2016-03-01:
@thunderdan After updating bootloader (to #719) I have done numerous firmware cycles and have not seen the spurious pwm output since. This may be a fix for your issue.

thunderdan, 2016-03-01:
Thanks.  I'll try tonight.

-- Dan Morgan --

thunderdan, 2016-03-12:
Update: A 2200uF electrolytic capacitor has solved my issue.  I'm attaching the cap to the PWM output pin header + and - pins.  The sparky2 makes it through initialization even with my "bad" battery and initialization completes setting the servos to neutral.  I've drained the battery down to 12V and I still can power-up just fine.  No brown outs during bench operation either.  I feel confident to go for a flight again now.

The servos still move during the power-up but it must be a pulse generated by the weak pull-ups mentioned by @tracernz.  Thanks @pug398 @mluessi @mlyle @tracernz

Back to the GPS for me :)

tracernz, 2016-03-12:
Good to hear @thunderdan. Thanks for reporting back. :+1:

closed 2016-03-12 by tracernz