d-ronin/dRonin #644: Reported: Linux deb package does not properly install udev rules

mlyle, 2016-02-10:
https://github.com/d-ronin/dRonin/issues/437#issuecomment-182450317

thunderdan, 2016-02-18:
I looked at this issue.  The debian installer puts the file at /lib/udev/rules.d/45-dronin.rules which is a valid location.  Kernel.org explains that files from this directory as well as /etc/udev/rules.d get collected and processed.

Maybe there is some confusion since the developer documentation specifically says to copy the rules file and place it at /etc/udev/rules.d/45-dronin-permissions.rules.

Hopefully this info saves you some time @mlyle .

Source of info:
https://www.kernel.org/pub/linux/utils/kernel/hotplug/udev/udev.html 

I'm running linux mint 17.  The results should the same on ubuntu.

mlyle, 2016-02-22:
@thunderdan-- thank you.  I'll double-check too, because it's better to be on the side of not losing an issue report, but this lets me prioritize it down.

mlyle, 2016-02-29:
Confirmed on my machine it installs `/lib/udev/rules.d/45-dronin.rules` -- thx again @thunderdan

closed 2016-02-29 by mlyle