d-ronin/dRonin #645: Centralize bank maps

dustin, 2016-02-10:
I dug through board-specific C++ code to generate a board map I could use for figuring out how people's timers are configured:

https://github.com/d-ronin/autotown/blob/master/static/banks.json

Ideally, we generate the code from a table that can be used elsewhere (since more than just the GCS where this code lives might want to use that data).

tracernz, 2016-02-10:
They are in the GCS board plugins, and in the board_hw_defs.c. I'm curious what this is useful for outside of there?

tracernz, 2016-02-10:
Oh, and also in the schematics :P. It's a board-specific implementation detail so not many things should care about it.

dustin, 2016-02-10:
It's useful because the actuator config sets frequency by bank and the outputs are by channel.  I want to be able to tell whether someone has a particular channel's frequency set up incorrect (or in the simple case, just what frequency they're using).

tracernz, 2016-02-10:
Hmm, we should try to represent the data in a different way so we don't need to deal with board specific stuff outside of places where it's not strictly necessary IMO.

dustin, 2016-02-11:
It's not entirely obvious how one could do this.  The channels are banked and organized by timer.  The timer can only have one frequency.  The user has to choose the frequency for the bank and has to do channel assignments.  We either have to reflect that, or do a ton of magic to make it work so that we can pretend each channel has its own frequency.

And that's really what I'm getting at.  What channel is the NE motor?  What's the frequency, resolution, and end points here?  If one of them is at 50Hz, should we have enough information to tell them how to fix it?

mlyle, 2018-04-20:
I think the data is represented in GCS well enough-- it's unfortunate it needs to be duplicated from flight but probably not worth having a central build mechanism for it.

closed 2018-04-20 by mlyle