d-ronin/dRonin #646: One report: Unable to connect to CC3D after flashing

mlyle, 2016-02-10:
Triple-verified correct procedure was tried.  Used an eraseflash firmware and then flashed back, to no success.  Windows 10 x64.

This is a placeholder for any similar results.

[image]

jpbarraca, 2016-02-12:
I have a CC3D and can test the process. The CC3D will be usable (can I flash it again) after this?

mlyle, 2016-02-12:
@jpbarraca yes.  On the other hand this is something that only happened once so you're not likely to repro it.

I talked to the user yesterday; he solved the issue with this procedure:

https://librepilot.atlassian.net/wiki/display/LPDOC/Troubleshooting

jpbarraca, 2016-02-12:
@mlyle I tried but I cannot reproduce this issue on OS X.

Maybe it is a good idea to add that info to our wiki.

mlyle, 2016-02-13:
@jpbarraca Yes-- it's only on Windows, and only happened to about 1 windows user out of 100.

matheusrgomes, 2016-02-28:
CDC Driver
The third-party INF does not contain digital signature information

Driver does not install = No connection to BrainFPV

mlyle, 2016-04-01:
Documented: https://github.com/d-ronin/dRonin/wiki/Windows-driver-troubleshooting

In errata.

closed 2016-04-01 by mlyle