d-ronin/dRonin #648: GCS crash when opening additional workspace

carbogit, 2016-02-10:
adding an additional workspace through Tools/Options/GCS/Workspaces and then going to that new workspace via its tab at the bottom of the screen crashes the GCS (Win 7 64 bit, Renatus release).

tracernz, 2016-02-11:
Confirmed on Linux. Tries to de-reference a null pointer at https://github.com/d-ronin/dRonin/blob/f801a2c5342ed4cfbc138480b5e5afbcedfb970e/ground/gcs/src/plugins/coreplugin/uavgadgetmanager/uavgadgetmanager.cpp#L135

mlyle, 2016-02-11:
Well, that's clever. :P

jpbarraca, 2016-02-12:
Confirmed on OS X

tracernz, 2016-02-12:
> Well, that's clever. :P

Well, it's cleverer on STM32 where 0x0 points at the base of flash :-P.

jpbarraca, 2016-02-21:
What is the usefulness of a user adding a new workspace? I digged a little in the interface and found nothing related.

tracernz, 2016-02-21:
Well the entire UI is fully configurable so you can setup whatever tabs you like in whatever way you like. I think probably 99% of people use the default setups provided though. I'm guessing this is related to #311 but could well be wrong because I didn't even read the diff.

tracernz, 2016-02-21:
Can confirm #311 is the cause of this crash.

mlyle, 2016-02-21:
OK.  Probably another instantiation case needs to be handled properly.

mlyle, 2016-02-29:
```
0   libCore_debug.dylib           	0x0000000110c6fa95 Core::UAVGadgetManager::modeChanged(Core::IMode*) + 53
1   libCore_debug.dylib           	0x0000000110cdbf49 Core::UAVGadgetManager::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 249
2   org.qt-project.QtCore         	0x0000000107be02e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
3   libCore_debug.dylib           	0x0000000110ce21cd Core::ModeManager::currentModeChanged(Core::IMode*) + 77
4   libCore_debug.dylib           	0x0000000110c99d15 Core::ModeManager::currentTabChanged(int) + 773
5   libCore_debug.dylib           	0x0000000110ce1f78 Core::ModeManager::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 408
6   org.qt-project.QtCore         	0x0000000107be02e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
7   org.qt-project.QtWidgets      	0x0000000106eedd96 QTabWidget::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 758
8   org.qt-project.QtCore         	0x0000000107be02e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
9   org.qt-project.QtWidgets      	0x0000000106ee5303 QTabBar::setCurrentIndex(int) + 387
10  org.qt-project.QtWidgets      	0x0000000106ee975c QTabBar::mouseReleaseEvent(QMouseEvent*) + 684
11  org.qt-project.QtWidgets      	0x0000000106d8efcc QWidget::event(QEvent*) + 1628
12  org.qt-project.QtWidgets      	0x0000000106ee6fd6 QTabBar::event(QEvent*) + 630
```

PR #722 (2016-03-01): gcs: uavgadgetmanager: handle adding new toplevel pane
fixes #648

[[image]](https://reviewable.io/reviews/d-ronin/dronin/722)
commit 0f4b111f (2016-03-01): gcs: uavgadgetmanager: handle adding new toplevel pane

fixes #648
closed 2016-03-01 by tracernz