d-ronin/dRonin #660: GPS: Move settings into own UAVO, Add GPS tab in ModuleSettings UI

tracernz, 2016-02-13:
Most of the settings aren't currently exposed in the UI. There are now enough to make it worth moving them into their own UAVO IMO and consequently free up a little RAM+Flash on F1.

mlyle, 2017-03-12:
No more F1; objection to closing WONTFIX?

tracernz, 2017-03-17:
I still think this is a good idea and make ModuleSettings less of a dogs breakfast. The UI is another issue.

mlyle, 2017-03-17:
Yah, there's a lot of refactor of settings uavos possible, though it will break existing configurations (unless we explicitly build migration code for, which there's not a great framework).  At the same time, priorities, you know?  If we open issues for things we're not actively tripping over, we could have a pretty-close-to-infinite set of cleanup issues open.

I think the bug database is most useful when it encompasses things that we're likely to consider reasonably high priorities.

tracernz, 2017-03-17:
Indeed. It's harder to keep track of what is actually done and what isn't though when things are closed/not-done. :wink: