d-ronin/dRonin #661: Input - RSSI field does not support all RX types

DTFUHF, 2016-02-13:
[image]

Missing: DSM, SUMD, RFM22B (though rfm22b already has its own rssi)

DTFUHF, 2016-02-13:
Somewhat related, PPM is limited to 12 channels, this can be a little restrictive. Sbus supports up to 18,  SUMD up to 32!

tracernz, 2016-02-13:
Is DSM useful? I don't think there's any way to make a Spektrum Rx or Tx spit out RSSI to a channel?

DTFUHF, 2016-02-13:
Heh, I guess not, maybe in a future RX

mlyle, 2016-04-07:
#856 made progress towards this.  I think the only thing needed really is to turn up the channel count.

mlyle, 2016-07-22:
This is fixed; we're stuck with channel counts for now...

closed 2016-07-22 by mlyle