d-ronin/dRonin #664: GCS:Libs: Upgrade to Eigen 3.3

tracernz, 2016-02-14:
When released (currently at beta http://eigen.tuxfamily.org). Previous versions use things that are deprecated in C++11, and cause compilation to fail with GCC 5.3 for @fujin.

fujin, 2016-02-14:
confirmed

jpahka, 2016-02-29:
Because of this in order to build current next, I need to add

QMAKE_CXXFLAGS += -Wno-error=deprecated-declarations

to the end of file ...ground/gcs/gcs.pri as per icee's suggestion.

mlyle, 2016-04-10:
Related: TauLabs/TauLabs#2240

Hopefully Eigen 3.3 will ship soon.

fujin, 2016-04-15:
you can cherry pick https://github.com/fujin/dronin/commit/34a79bff57af77b915dbea14242cafb5e940831dif it helps

tracernz, 2016-04-15:
Hopefully not far away https://listengine.tuxfamily.org/lists.tuxfamily.org/eigen/2016/04/msg00017.html

mlyle, 2016-08-06:
Eigen 3.3 beta-2 was released on July 26, 2016, so maybe there's some hope...

mlyle, 2016-09-01:
Hopefully soon.

mlyle, 2016-09-24:
Eigen 3.3 expected to ship next week.

tracernz, 2016-11-10:
Wowowowow http://eigen.tuxfamily.org/index.php?title=News:Eigen_3.3_released! :open_mouth:

PR #1456 (2016-11-15): GCS: Upgrade to Eigen 3.3
Fixes #664.

Also cleans up a bit of code in Calibration.
closed 2016-11-15 by mlyle