d-ronin/dRonin #665: TxPID for Damping and Noise sensitivity

ufoDziner, 2016-02-14:
Would be awesome to have the ability to adjust the Autotune Damping and Noise sensitivity using the TxPID module.  Thanks!

dustin, 2016-02-14:
This is another nice aspect of the concept of pushing AT computation into
the FC.  It's quite risky, but a really neat higher level usage of txpid.

mlyle, 2016-02-15:
Yah, though this one particularly makes me nervous.  It's a bit expensive of math to be doing in the middle of flight, and the tunes resulting may not be "continuous" / a small txpid deflection may do something wonky.

mluessi, 2016-02-16:
This would probably only make sense on flight controllers with an OSD, so the parameters could be adjusted and new PID gains computed/inspected without needing a computer. This would be nice, but we would need to make sure that the code on the flight controller and the GCS give the same results (the GCS code is probably using double precision math, so we would need to make sure it works with single precision, etc.).

droidicus, 2016-02-16:
Another option would be to lock the calculated PID values in place at arm, and require a dis-arm/re-arm cycle to load a new set in.

mlyle, 2016-10-05:
This probably won't play well with neo-autotune.  defer.

mlyle, 2017-02-06:
Unfortunately, this is the exact opposite direction from what we're thinking-- getting a great offline tune.  Closing WONTFIX

closed 2017-02-06 by mlyle