d-ronin/dRonin #668: GCS: Importing UAV settings sets all UAVOs

mluessi, 2016-02-16:
When importing an .uav file in the GCS, all objects present in the file are set, even if they are not selected to be saved to flash. This should not happen, as settings are changed without informing the user and attempting to fly without first restarting the flight controller (which will cause the correct settings to be read from flash) can cause a crash/unintended behaviour etc.

jpbarraca, 2016-02-16:
I have this ready but PR #655 must be accepted (or denied) first as it will change some of the same code.

tracernz, 2016-02-16:
@jpbarraca, if you could take care of the minor formatting issues in #655 and I'll test+merge after work tonight so you can get on this one. :-)

jpbarraca, 2016-02-16:
@tracernz I think everything is already fixed in #655

tracernz, 2016-02-16:
Sorry, missed the last commit, thanks! I'll be home in about 8 hours.

PR #669 (2016-02-22): GCS: Do not automatically apply UAV settings with import
This will import all settings to a temporary object, but nothing will be modified in the FC without user intervention. The apply button will make settings temporarily active (apply to ram), while the save button will save settings to the persistent memory (flash)

Fixes #668

[[image]](https://reviewable.io/reviews/d-ronin/dronin/669)
closed 2016-02-22 by mlyle