d-ronin/dRonin #670: Flight - Make ef_target cmds no longer work in Windows Development Environment

jihlein, 2016-02-17:
After rebasing PR #403 to the latest next, make ef_target fails as follows:

recipe for target 'c:/dronon/build/ef_aq32/eq_aq32.fw_pre.pad' failed

All targets fail in the same manner.  This works on Linux, not sure about OSX.

mlyle, 2016-02-18:
Do other targets fail the same way, or just aq32?

It appears to work for me (and the automated build environment).  I'm not able to try the branch.

I'm not able to try this thoroughly while on my trip, unfortunately.  Could AQ32 be out of space / nearly out of space in a partition?  

(Incidentally I think I recently noticed that AQ32 just has a single logfs arena which is an incorrect configuration).

jihlein, 2016-02-18:
All targets are failing the same way, but I'm on the road and using a different laptop for building, maybe the environment isn't setup quite right.  I'll check this when I get home on my normal development laptop, along with the logfs arena.

jihlein, 2016-02-20:
Must be an issue with that particular laptop, my primary development laptop at home works fine.

Closing.

closed 2016-02-20 by jihlein