d-ronin/dRonin #679: FrSky S.Port Telemetry broken on next

droidicus, 2016-02-23:
FrSky S.Port Telemetry is broken on next with the following configuration:
Brain FC
FlxPort - S.Port telemetry
MainPort - S.Bus serial
RXPort - disabled

@jihlein it looks like this problem is corrected by reverting #403, not sure why

mluessi, 2016-02-23:
@jihlein I just had a look at `pios_board.c` for brain. It seems like #403 removed the inverter configuration. Was this intentional? C.f.

https://github.com/d-ronin/dRonin/pull/403/files#diff-2ff060fb48c128c0340a34d5d954f6dfL450

jihlein, 2016-02-23:
Can't read the diffs too well on my phone. I'll check into this later after I'm off the plane.

mlyle, 2016-02-23:
A little regression is normal with a big changeset like this-- thanks @jihlein.  We weren't likely to hear about it while it sat in a PR with sparse test.

jihlein, 2016-02-23:
@mluessi, for the Brain, the inverter pin is defined in lines 577 - 590 of board_hw_defs.c.  That structure is passed to PIOS_HAL_ConfigurePort in line 444 of pios_board.c.  Lines 420-434 of pios_hal.c configure the pin based on whether or not SBUS is the selected port configuration.  @pug398 tested both sbus and sbus ni on the Brain and found them to work, but I'm not sure on which port.  I'm assuming main.

@droidicus, FrSky Sport Telem was reported to be working on Sparky, but's that an F3 target.  Can you confirm if anything works on the Brain FlxPort?  The code is pretty much the same and I don't see anything obvious at first glance.

pug398, 2016-02-23:
Yes I tested sbus and sbusni on Brain main port. If the question is with frsky smart port telemetry I can test later on. I have an sport inverter board for F4 laying somewhere.

jihlein, 2016-02-23:
@pug398, could you also give sbusni a test on the flex port if it's not to much trouble?

pug398, 2016-02-23:
Will do. It is no trouble. You are the one that did all the hard work :)

pug398, 2016-02-23:
With Brain/next I have working:
sbus on main
sbusni on main
gps on main
sbusni on flexi
gps on flexi
I have been unable to get s-port telemetry to work on flexi port.

jihlein, 2016-02-24:
Interesting.  So flexi port is configuring some functions correctly.  It sound like an F4 target issue, as I recalI that you had s-port working on Sparky, correct?

Wonder if it has anything to do with the recent F4 one wire change.  I'll have to look into that.

pug398, 2016-02-24:
I am going to go back and verify sparky/next with this rx because I am unable to get s-port telem to work on any F4 now. My inverter uses rx/tx/sport connections.

pug398, 2016-02-24:
s-port is working on sparky with current next. Will see if I can get one of the F4 to work.

jihlein, 2016-02-24:
It won't work on any F4 right now.  

The F4 one wire changes is throwing the F4 uart into one wire mode and the current crop of inverters that use RX and TX pins won't function.  The PIOS_HAL code sets the one wire flag for F3 targets.  Previously that flag was unused on the F4 and had no effect on the F4 uart.  Now it does.  Need to rework something here.

You might be able to hook both inputs of the inverter to the TX pin.  I'm not 100% sure as I don't have the inverter schematic handy.

mlyle, 2016-02-24:
That probably won't work :/  Ah, so it's an interaction between those two changesets.

Since the other one is unused right now we could consider reverting it.

@mluessi ?

tracernz, 2016-02-24:
So why does it work on F3 then? It has always done one wire right?

mlyle, 2016-02-24:
It breaks F4 because it's enabling one-wire on F4 now, but can't enable inversion (there's no inversion on F4).  Existing peoples' inverters don't work with one-wire.  Combination of #403 and #672.

mlyle, 2016-02-24:
(But F3 has always done inversion + one wire to be glueless)

tracernz, 2016-02-24:
Oh. Kind of a tricky issue since it's theoretically better but the hardware people already have sucks. We will be using one wire on F4 soon with the work m_thread and myself are doing so it would be a shame to revert it because of poor choices by FrSky, leaving better protocols to deal with it.

mlyle, 2016-02-24:
Bidirectional inversion is problematic, though.

tracernz, 2016-02-24:
So let the FrSky code deal with F1/F4 as special cases (i.e. disable 1-wire) as necessary rather than breaking 1-wire support for everyone else.

mlyle, 2016-02-24:
Sure, that's a good longterm solution.

But since nothing uses F4 onewire yet, it could be an OK hack for release to just revert #672.

Don't really care what we do in the short term, but it needs fixin'.

jihlein, 2016-02-24:
The following could be an interm fix, would also require modifying pios_config.h file for the F3 targets.

`	case HWSHARED_PORTTYPES_FRSKYSPORTTELEMETRY:
#if defined(PIOS_INCLUDE_FRSKY_SPORT_TELEMETRY)
		PIOS_HAL_ConfigureCom(usart_port_cfg, &usart_port_params, PIOS_COM_FRSKYSPORT_RX_BUF_LEN, PIOS_COM_FRSKYSPORT_TX_BUF_LEN, com_driver, &port_driver_id);
		target = &pios_com_frsky_sport_id;
#endif /* PIOS_INCLUDE_FRSKY_SPORT_TELEMETRY */

#if defined(PIOS_INCLUDE_FRSKY_SPORT_TELEMETRYF3)
		usart_port_params.rx_invert   = true;
		usart_port_params.tx_invert   = true;
		usart_port_params.single_wire = true;

		PIOS_HAL_ConfigureCom(usart_port_cfg, &usart_port_params, PIOS_COM_FRSKYSPORT_RX_BUF_LEN, PIOS_COM_FRSKYSPORT_TX_BUF_LEN, com_driver, &port_driver_id);
		target = &pios_com_frsky_sport_id;
#endif /* PIOS_INCLUDE_FRSKY_SPORT_TELEMETRYF3 */
		break;
`

jihlein, 2016-02-24:
The code below would work, would have to modify pios_config.h for the F3 targets too.

```
	case HWSHARED_PORTTYPES_FRSKYSPORTTELEMETRY:
#if defined(PIOS_INCLUDE_FRSKY_SPORT_TELEMETRY)
		PIOS_HAL_ConfigureCom(usart_port_cfg, &usart_port_params, PIOS_COM_FRSKYSPORT_RX_BUF_LEN, PIOS_COM_FRSKYSPORT_TX_BUF_LEN, com_driver, &port_driver_id);
		target = &pios_com_frsky_sport_id;
#endif /* PIOS_INCLUDE_FRSKY_SPORT_TELEMETRY */

#if defined(PIOS_INCLUDE_FRSKY_SPORT_TELEMETRYF3)
		usart_port_params.rx_invert   = true;
		usart_port_params.tx_invert   = true;
		usart_port_params.single_wire = true;

		PIOS_HAL_ConfigureCom(usart_port_cfg, &usart_port_params, PIOS_COM_FRSKYSPORT_RX_BUF_LEN, PIOS_COM_FRSKYSPORT_TX_BUF_LEN, com_driver, &port_driver_id);
		target = &pios_com_frsky_sport_id;
#endif /* PIOS_INCLUDE_FRSKY_SPORT_TELEMETRYF3 */
		break;
```

mlyle, 2016-02-24:
Yah, I'm OK with that too.

jihlein, 2016-02-24:
There's few other small mods required too, nothing significant.  Not sure I want to cut a PR on this computer, will need to wait until I get home unless someone else wants to tackle it.

tracernz, 2016-02-24:
The only thing I don't like about that is that F3 is the normal one, F1/4 are the special cases but that's being very pedantic :-P.

jihlein, 2016-02-24:
Easy enough......

PR #700 (2016-02-26): Flight: Fix broken Frsky Sport Telem on Next.
This PR corrects the conflict between PR #403 and PR #672, addressing Issue #679.

Can somebody with Frsky Sport telem and F4 test this?

[[image]](https://reviewable.io/reviews/d-ronin/dronin/700)

mpl edit: fixes #679 (for linkage)
closed 2016-02-26 by mlyle