d-ronin/dRonin #699: Actuator Alarms are a bit noisy

fujin, 2016-02-25:
In testing of LPS/ACM, while disarmed, I get 'ACTUATOR' hard on the screen until throttle is moved above zero (regardless of arm state)

I also get a lot of 'EVENT' followed by 'ACTUATOR' and sometimes 'EVENT: ACTUATOR'? Not sure.

OhmEye, 2016-02-29:
On current next I have ACTUATOR warning on brain OSD, and in GCS the servo input block is yellow with no tooltip explaining why, and on output page the motor test does not allow, with reason "the actuator module is in an error state. This can also occur because there are no inputs. Please fix these before testing outputs"

I see value in alarming on clipping, but don't see value in alarm for normal operation, and am confused why anything actuator related should be alarming when in a disarmed state.

fujin, 2016-02-29:
I didn't pick up on the inability to do motor test on outputs page :( I will try to confirm.

fujin, 2016-02-29:
From IRC:

```
12:02:53 <icee> ah
12:02:54 <icee>         mbox.setText(QString(tr("The actuator module is in an error state.  This can also occur because there are no inputs.  Please fix these before testing outputs.")));
12:02:54 <fujin> finally going to test dR MSP -> MSWOSD
12:02:57 <icee> this thing
12:02:58 <icee> OK, you're right
12:03:02 <icee> let me fix that
12:03:04 <fujin> Yes, that's the one he was seeing.
```

fujin, 2016-03-28:
Will test.

PR #835 (2016-03-30): actuator: only alarm when conditions met for stabilization
Fixes #699

---
This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/835)
commit a6546dc7 (2016-03-30): actuator: only alarm when conditions met for stabilization

Fixes #699
closed 2016-03-30 by tracernz