d-ronin/dRonin #701: MWRateSettings object save timeout during smart save after autotune

fujin, 2016-02-26:
It doesn't seem like these are updated anyway so not sure why they are saving. Current next

```
18:36:35[DEBUG]ConfigGadgetWidget onAutopilotConnect
18:38:00[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "StabilizationSettings"
18:38:00[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "ActuatorSettings"
18:38:00[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "MWRateSettings"
18:38:01[DEBUG]RESULT "All operations finished successfully" () () ()
18:38:01[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "StabilizationSettings"
18:38:01[DEBUG][smartsavebutton.cpp] Save request for object "StabilizationSettings"
18:38:01[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "ActuatorSettings"
18:38:01[DEBUG][smartsavebutton.cpp] Save request for object "ActuatorSettings"
18:38:01[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "MWRateSettings"
18:38:01[DEBUG][smartsavebutton.cpp] Save request for object "MWRateSettings"
18:38:03[DEBUG][smartsavebutton.cpp] Saving timeout for object "MWRateSettings"
18:38:03[DEBUG][smartsavebutton.cpp] failed to save: "MWRateSettings"
18:38:03[DEBUG]RESULT "Objects not saved:\nMWRateSettings\n" () ("MWRateSettings") ()
18:38:08[DEBUG]Automatically opening device
18:38:08[DEBUG]ConnectionManager::devChanged autoconnected USB device
```

tracernz, 2016-02-26:
Note: Deprecated in post-tanto.

mlyle, 2016-02-26:
Just more evidence for flaky telemetry IMO.

mlyle, 2016-02-26:
Just to confirm-- current next, and not the UI-change-branch?  (Where we're saving one more object, actuatorsettings, on this tab).

mlyle, 2016-06-25:
Dup #641 or fixed.

closed 2016-06-25 by mlyle