d-ronin/dRonin #702: Afromini Amaze fails to be detected properly

dustin, 2016-02-26:
Afromini Amaze is coming up with `hse_value == 8000000`.  AFAICT, the code to figure out the `hse_value` is just about the same as baseflight.  Baseflight seems to also detect this as ≠ rev5, but manages to get i2c working.

If I force it to be considered a rev 5, it works fine.

(note:  I also tried next + one-true-mpu and had the same thing, just to be safe)

I'm sure I can fight with the baseflight code a bit more to figure this out.

tracernz, 2016-02-26:
STMicroelectronics code detects the HSE freq AFAIK.

dustin, 2016-02-26:
Right.  There's an odd difference between baseflight and dRonin's copies of the libraries, but it has no effect, AFAICT.  `hse_value` is a red herring.  I've got more work to do here, but I want to use one of these things, and it works if I use the different i2c config, so it shouldn't be *that* hard to make it work in general.

DTFUHF, 2016-02-27:
Since close to nobody has rev4 / afrominiv2, the mpuint pin is always on PC13, any other configuration is too old to even matter. The "12MHz detect pin" should only be used to detect which oscillator is used

dustin, 2016-02-27:
I've got a v2.  It'd be awesome if I could make that work, but haven't got
to it yet.   I wonder how hard it'd be to stuff a hardware option somewhere
for this.

DTFUHF, 2016-02-27:
I think even rev2 has mpuint on PC13 (maybe you can verify) making rev4 support even less needed ¯\_(ツ)_/¯

mlyle, 2016-02-27:
Wish we could estimate how many R4s there are in our userbase...

dustin, 2016-02-27:
I think I've got an OK workaround.

closed 2016-02-27 by mlyle

tracernz, 2016-02-28:
>Wish we could estimate how many R4s there are in our userbase...

Well, I have this in the mpu branch, and the first half (runtime setting of revision) is already merged..

tracernz, 2016-02-28:
Would be trivial to add to current code since we already detect Rev.

mlyle, 2016-02-28:
Yes, we should add it, but that's not very informative for what we should do in the short term.

PR #712 (2016-02-27): naze32: configurable interrupt port for mpu6050
The port moves about among the various naze incarnations and isn't as
simple to detect as "is this a rev 5?"  Afromini Amaze is effectively a
rev 4, but has the MPU on port C.

This allows the user to override that when the default isn't working
correctly.  My Amaze works now.

Closes #702 

[image]

[[image]](https://reviewable.io/reviews/d-ronin/dronin/712)
commit 0011dd1e (2016-02-27): naze32: configurable interrupt pin for mpu6050

The pin moves about among the various naze incarnations and isn't as
simple to detect as "is this a rev 5?"  Afromini Amaze is effectively a
rev 4, but has the MPU on port C.

This allows the user to override that when the default isn't working
correctly.  My Amaze works now.

Closes #702