d-ronin/dRonin #705: ship android GCS

mlyle, 2016-02-26:
* [ ] Building
* [ ] Jenkins build
* [ ] Some branding

Lots of people want this-- I was surprised.

dustin, 2016-02-26:
Autotune?
Phone home?

mlyle, 2016-02-26:
Not interested in spending a lot of time on this, just restoring basic functionality for the release.

tracernz, 2016-02-26:
If we could find a Java/Android dev we could do some cool stuff. As is we just don't have time.

mlyle, 2016-02-26:
Yah, I think our best bet long term is smushing down the normal GCS onto phones/tablets.

mlyle, 2016-02-28:
Branding -- ic_icon.png, ic_logo.png, in ldpi, mdpi, hdpi, xhdpi (androidgcs/res/drawable-...)

mlyle, 2016-03-24:
Good enough for Tanto.  Haven't addressed signing.

mlyle, 2016-04-12:
Will watch the download counts and problem reports with AndroidGCS to decide how important this stuff is.

ragd011, 2016-05-13:
I've been downloading them and trying to use them.  I was hoping that I could adjust tuning settings in the field.   I was able to connect but couldn't seem to do much.  A little while back I tried to mess with the source but it was in an older environment so migrating it and tweaking it is not a good first project :-)  So anyway, what is the AGCS useful for in its current state?

PR #1326 (2016-09-12): androidgcs: enhanced support -- maps, arch support, uavo collection
Fixes #705
closed 2016-09-12 by tracernz