d-ronin/dRonin #710: Kitchensink shake-out branch bookkeeping

mlyle, 2016-02-27:
* [ ] c4d607c Merge remote-tracking branch 'pr/708' into mpl-kitchensink #708 sim compile on windows
* [ ] 85fe483 Merge remote-tracking branch 'pr/707' into mpl-kitchensink #707 althold intergal
* [ ] d981cbb Merge remote-tracking branch 'pr/704' into mpl-kitchensink #704 remove flash "OTA"
* [ ] c2e0f51 Merge remote-tracking branch 'pr/697' into mpl-kitchensink #697 gcs hangtime settings
* [ ] 1ca47e4 Merge remote-tracking branch 'pr/696' into mpl-kitchensink #696 setupwiz always requests reboot
* [ ] 522d4ae Merge remote-tracking branch 'pr/693' into mpl-kitchensink #693 gcs CWD / save dialog
* [ ] 3d474f0 Merge remote-tracking branch 'pr/691' into mpl-kitchensink #691 failsafe disarm timer
* [ ] e37d1b8 Merge remote-tracking branch 'pr/690' into mpl-kitchensink #690 shortcutsettings remove
* [ ] 6a4ff72 Merge remote-tracking branch 'pr/676' into mpl-kitchensink #676 lowpowerstab
* [ ] 779f199 Merge remote-tracking branch 'pr/617' into mpl-kitchensink #617 revo flexio uart
* [ ] e89ddff Merge remote-tracking branch 'pr/576' into mpl-kitchensink #576 cellcountcalc

mlyle, 2016-02-27:
artifacts will show up here: http://jar.lyle.org:8080/job/dronin/188/

mlyle, 2016-02-27:
Kitchensink sequence 1 will be on lux; hope to evaluate 576, 676, 690, 691, 693, 696, 697, 704, 711.

mlyle, 2016-02-27:
Updated kitchensink for failsafe change in 676.

artifacts will show up here: http://jar.lyle.org:8080/job/dronin/192/

pug398, 2016-02-28:
On kitchen2, sparky2 has a constant actuator warning that is not on next 5fde.

mlyle, 2016-02-28:
@pug398 -- bug #699 -- if we're unable to meet roll/pitch command because of clipping, we now actuator alarm.  Unfortunately this includes the case when we can't because there's no throttle.

Wow, you're brave flying this :P  

Low power stabilization is fun on small craft.  Have not dared to try it on large ones.

mlyle, 2016-02-29:
#690 / #691 are the last thigns I hope are mergeable form this test pass, and I think they'll be reviewed soon.  So closing this bookkeeping issue for this kitchensink.

closed 2016-02-29 by mlyle