d-ronin/dRonin #716: DSM binding on inverter targets (Sparky2, Quanton, Brain, CC3D, Revo) sucks

mlyle, 2016-02-29:
* Some targets silently squash binding pulses when DSM is on an inverter port (S2, Revo)
* Some targets drive the external-inverter port and busfight (Quanton, Brain, CC3D?)
* Revo erroneously configures DSM to an inverter port by default.

@tracernz -- does this cover it?

tracernz, 2016-02-29:
@mlyle Yeah I think so.

tracernz, 2016-02-29:
The Revo is really a separate issue IMO and is a regression rather than a pre-existing bug.

henn1001, 2016-02-29:
Yeah sorry about the revo - i didn´t quite get the logic behind the module :/

jihlein, 2016-02-29:
In the FWIW category, an easy fix is to remove DSM support on ports with hardware inverters.  That's the path I took on the AQ32 target, no DSM on uart3.  Need to review targets in question, don't know if it would create other difficulties.

mlyle, 2016-02-29:
We could do that, but if you have an already bound satellite it's a little limiting.  It may be the better of choices, as people will get confused about thigns not binding.

OTOH binding works so damn poorly :/

tracernz, 2016-02-29:
Might be cleanest just to remove DSM on those ports, although not ideal if it happens to be a dedicated Rx port..

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.  Sorry :(

PR #918 (2016-04-15): flight: don't busfight on DSM binding
When there's an inverter/xor present, it's essential that we not drive
its output.  Fixes #716, makes Quanton, Brain, CC3D consistent with S2
and revo.
commit 2acb4406 (2016-04-15): flight: don't busfight on DSM binding

When there's an inverter/xor present, it's essential that we not drive
its output.  Fixes #716, makes Quanton, Brain, CC3D consistent with S2
and revo.
closed 2016-04-15 by tracernz