d-ronin/dRonin #725: iOS GCS

tgreer, 2016-03-01:
Opening issue to discuss iOS GCS. There are solutions out there to create a TCP<>Serial bridge, https://github.com/beckdac/ESP8266-transparent-bridge as an example uses a $5 component.

I understand that computer GCS currently uses UDP, is there a reason this can't be TCP?

WZ9V, 2016-03-01:
I'd be interested in this also.

mluessi, 2016-03-01:
AFAIK, the GPL is incompatible with the app-store TOS, so the iOS GCS would need have to have a different licence and therefore would probably have to written from scratch.

tgreer, 2016-03-01:
@mluessi The internet agrees with you. I think it'd need to be written from scratch anyways...

mluessi, 2016-03-01:
@tgreer yeah, the license thing is unfortunate. One possibility could be to build a Qt / QML based app, which would have the advantage that it would also work on Android. In addition, we could probably license the app such that components from it could be used in the normal GCS (but not the other way around).

The question is, what is the scope of the app? Recreate the whole GCS, i.e., board configuration, calibration, etc, or do we want an app for doing navigation flights (setting waypoints etc)? The latter may be more easily accomplished by improving our MAVLINK support, so existing apps for APM could be used.

tgreer, 2016-03-01:
I'd like it to be as featured as the existing Android one... Granted if
we're starting from scratch the remit can be whatever the dev decides... I
would like it to be fully functioning... but I can dream, right ;)

WZ9V, 2016-03-01:
I'm willing to help but I'm currently in over my head. Working on a BLHELI GUI in Qt that I'm hoping can be the basis for adding that functionality to the GCS.  Keep in mind I'm also learning Qt at the same time ;-)

sonicxx, 2016-10-10:
There's also the possibility of releasing without the App Store as an enterprise app. Also there is cydia for jailbreakers.

sonicxx, 2016-10-11:
Apparently it had something to do with Mozilla software licensing? As to why they feel they are in full compliance.

mlyle, 2016-10-11:
Yes, if it's licensed under a license that doesn't have the "no further restrictions" clause all is good.

Unfortunately it requires consent of license holders to do this.