d-ronin/dRonin #727: Something missing in QT linux packaging

mlyle, 2016-03-01:
```Cannot mix incompatible Qt library (version 0x50201) with this library (version 0x50501)```

tracernz, 2016-03-01:
Ahhhh, can't seem to repro this anymore :(.

tracernz, 2016-03-01:
Hmmm, managed to repro it with PR artifacts though, problem children:
```
mike@mike-desktop:~/Desktop/package-20160229-d9f0b20d/gcs_linux_amd64-20160229-d9f0b20d$ grep "/usr/lib/x86_64-linux-gnu/libQt5" ./bin/lolz.txt 
[pid 25385] open("/usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5", O_RDONLY|O_CLOEXEC) = 8
[pid 25385] open("/usr/lib/x86_64-linux-gnu/libQt5SerialPort.so.5", O_RDONLY|O_CLOEXEC) = 8
[pid 25385] open("/usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5", O_RDONLY|O_CLOEXEC) = 8
[pid 25385] open("/usr/lib/x86_64-linux-gnu/libQt5Multimedia.so.5", O_RDONLY|O_CLOEXEC) = 8
[pid 25385] open("/usr/lib/x86_64-linux-gnu/libQt5Test.so.5", O_RDONLY|O_CLOEXEC) = 8
```

tracernz, 2016-03-01:
Confirmed it is fixed by copying these libs into libs/dr.

tracernz, 2016-03-01:
Ooops, I missed https://github.com/d-ronin/dRonin/blob/next/package/Makefile.linux#L53 in #680.

PR #728 (2016-03-01): Packaging: Fix broken lib deps on Linux, s/tau/dronin more
Fixes #727

[[image]](https://reviewable.io/reviews/d-ronin/dronin/728)
commit 22cf19b8 (2016-03-01): Package: Fix broken deps

Fixes #727
closed 2016-03-01 by mlyle