d-ronin/dRonin #734: PIOS HAL S.BUS: Possible to use external inverter?

fujin, 2016-03-02:
I've been backtracking through the PIOS_HAL branches to try to figure out how to setup a pios_sbus_cfg or perhaps use NULL? [by the looks](https://github.com/d-ronin/dRonin/blob/next/flight/PiOS/Common/pios_hal.c#L409-L422) in order to use an external inverter for s.bus with a board that I have.

Do we need a PIOS_SHAREDPORT_SBUS_EXTERNALINVERTER shared port type, now?

I can't figure this out, would appreciate any pointers. Please consider closing this issue, if it is possible and can be shown how it is done.

fujin, 2016-03-02:
boop @mluessi @jihlein

henn1001, 2016-03-02:
Not sure if this answers your question but you need a sbus_cfg like this: https://github.com/d-ronin/dRonin/blob/next/flight/targets/brain/board-info/board_hw_defs.c#L577
and then add it to the hal configuration like so : https://github.com/d-ronin/dRonin/blob/next/flight/targets/brain/fw/pios_board.c#L444

fujin, 2016-03-02:
cool, I thought that may have been the case but can't seem to figure out the structures required, will have to get @mluessi to help. thanks @henn1001!

closed 2016-03-02 by fujin

jihlein, 2016-03-02:
You don't need to define an sbus_aux_cfg structure, that's the pin to turn on the external inverter.  In the PIOS_HAL_ConfigurePort call, set the sbus_cfg argument to NULL.  Then simply use the S.Bus Non Inverted mode and the non inverted signals will be on the selected port pins.  Connect those to your external inverter.

henn1001, 2016-03-02:
Ah. Sorry i thought external inverter = Xor chip on the FC board. (internal inverter = f3 inversion).
Then you obviously don´t need that config with s.bus non inverted mode..

jihlein, 2016-03-02:
Actually that's a great question, as I took it to mean an inverter that is external to the board.  Must still be asleep this morning.....

Let's phase it this way:

If the inverter is external to the chip (F1/F4) and is physically on the FC board and needs to be switched on, what @henn1001 said above is correct.  Setup the pin with the sbus_aux_cfg structure, send that structure to PIOS_HAL_ConfigurePort, and use the S.Bus mode.

Sorry for the confusion. :) :)

fujin, 2016-03-02:
So in this particular case the external inverter is not an Xor chip, it is literally an external inverter (two resistors: 1x 4k7, 1x 10k & and a bc547 NPN.) thanks to both of you. I'll try the sbus_cfg argument to null + sbus-ni.