d-ronin/dRonin #736: PIOS_HAL_ConfigurePort shared port requiring serial comms, missing required usart parameters should be compile-time assertion/error?

fujin, 2016-03-02:
So I've just been diagnosing an issue where the receiverport for a board that I was using didn't have a usart cfg defined, yet was configured in UAVOs as being able to run sbus / sbus-ni.

It builds fine with the usart cfg set to NULL, as well as without the usart com driver, but when you boot the board, RC Input goes into 'critical' (driver failed to initialize for one of the input channels) with systemsalarm->mc saying `Settings`.

This stumped me for a bit. Adding the usart cfg for one of the other serial ports that is working and adding the pios usart com driver made it so RC Input came up in yellow state (no Rx) which tells me I'm on the right path.

It feels like to me if a given hw port is a shared port that has some configurations which require a usart to be present (i.e. s.bus, s.bus-ni), that it should fail to compile if no usart is given to hal_pios_configureport. Thoughts? cc/ @jihlein

fujin, 2016-03-02:
[Relevant changeset](https://github.com/fujin/dronin/commit/6578020b4b27e80ace035c07e422cdb8c37da8b7)

mlyle, 2016-03-02:
I don't know how to make that fail to compile in C.

fujin, 2016-03-02:
Can we at least generate a better error then? It seems like we ought to be able to infer that the required parts (a config, and the com driver) are there for shared hw ports requiring serial comms, right?

tracernz, 2016-03-02:
It seems like the alarm works fine and it fails as expected? I think we should only have specific indication for errors from user land.

fujin, 2016-03-02:
At least I'll know what to look out for now :)

closed 2016-03-02 by fujin