d-ronin/dRonin #745: gcs: don't allow selecting invalid flight modes

jhitesma, 2016-03-04:
Would be nice if we could disable selection of flight mode options that would cause a configuration error.  Specific example is that if the autotune module isn't enabled don't allow selecting autotune as a flight mode.

I make the mistake fairly often of adding autotune flight mode but forgetting to enable the module.   I know to expect the error and what it means, however it can be very frustrating to new users.

dustin, 2016-03-04:
I was actually thinking about this the other way around.  Don't have a checkbox for enabling autotune in the autotune page, just turn it on if it's selected as a mode, and turn it off if it's not.

jhitesma, 2016-03-04:
That would be really slick, hadn't considered that it may be possible to do that.

I was thinking this may be useful for other situations as well, like not allowing "manual" when vehicle type is a multirotor.  Those are the only two situations I can think of off the top of my head, but seems like it could potentially be applicable to more than just autotune.

mlyle, 2016-03-04:
https://github.com/d-ronin/dRonin/pull/652#issuecomment-183130591

I mostly want to solve something like this by replacing the vehicle setup wizard and the main sysalarms display with a "config checklist"-- something that looks at config objects and alarms and tells you what tasks need to be done still, with buttons to take you to those places.