d-ronin/dRonin #747: dRonin GCS intermittently fails to load 'ManualControl' object on CC3d on OSX

mlyle, 2016-03-04:
Probably relates to #654.

Clicking 'fetch' in the uavo browser is not doing it either.  Stays gray.

Have had this happen twice.  Now have tried 35 times in a row without repro on any object (10 were with logging).

dustin, 2016-03-06:
I just autotuned my 550 tri with cc3d in my basement.  When I finished, I plugged it back into the GCS and saw that it had some numbers.  I recorded the tune and then noticed it had [zero tau](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgJjw8AgM).  I had a GCS crash so it restarted.  I tried explicitly restarting as well and saw the same thing.  Some values were showing, but not all.

I unplugged the USB cable and plugged it back in and [the rest of the numbers showed up](http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgNL_ugsM).

mlyle, 2016-03-28:
Provisionally deferring this.

Also deferring all other telemetry bugs.

I have run 20 test cycles in the past couple of days with no repros on a fairly complicated CC3D configuration.  Also have not seen this since that first day.

Will keep an eye out.

mlyle, 2016-09-08:
I think this one is unhelpful given the other telemetry bugs.  Maybe I had mismatched GCS, maybe it was one of the other telemetry bugs manifesting a different way.

closed 2016-09-08 by mlyle