d-ronin/dRonin #748: Excessive event alarms

mlyle, 2016-03-05:
Happened on Tau, but now it happens even more I think.  Need to run down latency and bursts of excessive processing everywhere.

mlyle, 2016-03-05:
It's worth noting that it sure seems like toggling other alarms makes EVENT prone to stack up itself.

mlyle, 2018-03-16:
Been thinkin' -- is this telemetry?

tracernz, 2018-03-16:
Things like manualcontrol etc. set to onchange generate more updates than the link can handle I think? Should have a throttled onchange.

mlyle, 2018-03-16:
we have a throttled onchange.

tracernz, 2018-03-16:
How does it handle the tricky bit? If it always sends the last state in all cases then we could replace some onchange with throttled.

mlyle, 2018-03-16:
It doesn't send the last event in al lcases, but everything high frequency has a throttle value.

We don't know that this is telemetry.  It doesn't entirely make sense that it is (e.g. we used to see EVENT on screen in the OSD in flight before we masked it)

tracernz, 2018-03-16:
So, it puts a pointer in a UAVO somewhere with who triggered it, but IIRC that pointer changes a bit.

mlyle, 2018-03-16:
It puts the uavo id.  I'll check it out ;)  Just wanted to remind myself here to look at telemetry again.

mlyle, 2018-05-09:
I am seeing this much less lately....