d-ronin/dRonin #761: naze32: boot fault with autotune enabled

dustin, 2016-03-06:
Going through upgrades today (`next`), I have one with a pretty bare config (ppm on naze32) that won't boot correctly with autotune enabled.

This particular aircraft took my previous tune just fine, but this is going to be unfortunate for those without working tunes.

dustin, 2016-03-06:
Right as I did this, #762 shows up and likely fixes it.

dustin, 2016-03-07:
I built #766 and this seemed to make naze autotune happy.

dustin, 2016-03-07:
Marking this as fixed with The Great Merging of #766

closed 2016-03-07 by dustin

Zucky16, 2016-08-01:
I was about to create a new issue, when I found this one that is exactly what I was experiencing.  After saving/reloading/rebooting/power cycling I came to a possible conclusion.
I would have this boot problem as well as a failed autotune save when the naze32 rev 5 board I am testing on had MSP set instead of Telemetry as I had configured an MicroMinimOSD for use, and set the serial comms to MSP.
If I left that on MSP and tried to save the config with the Autotune module enabled, boot would fail and no Autotune module enabled.
Switch this back to Telemetry, save, reboot, reconnect, enable Autotune module, save, don't disconnect but simply remove all power from board, reconnect USB, reconnect via COM, all set and all green across the board for the board.  So it would seem that Autotune does not like naze32 rev 5 with MSP enabled, which is not really a big deal, as Autotune is just there for setup, but it was a pain to figure out what was causing the problem with enabling Autotune.

Let me know if another issue should be opened or if this one is the same.

tracernz, 2016-08-01:
When MSP in enabled, Telemetry is also enabled as a fallback (so when you connect GCS to your MSP port, the MSP module will detect UAVTalk packets, and hand control over to Telemetry, this allows the RX/TX pins in the middle of the Naze board to be used for OSD etc. without locking you out of GCS via USB). The result of this is that it uses a lot more memory, and when autotune is enabled too you run out of memory => boot failure.

tracernz, 2016-08-01:
I have opened #1163. This has been discussed a bit in the past and some work done toward it but no actual encompassing issue was opened before.

Zucky16, 2016-08-01:
Creating repetative new issues just slows down your progress, I spent a good 25 mins looking around the issues list to find something similar so as to not duplicate an issue, at least to the best of my abilities.

#1163 looks perfect, a notification of some kind would be very helpful to others.  Your explaination is excellent and makes complete sense to me.

P.S. Great work on this project, I'm loving it! :heart: