d-ronin/dRonin #763: Sparky1 may or may not be able to run at 1kHz (with an unknown set of modules activated)

fujin, 2016-03-06:
Per nog on IRC.

```
08:57:59 <nog> bbumped mpu up to 1000 and im getting cpu overload state
```

dustin, 2016-03-06:
I had the same issue on mine.

mlyle, 2016-03-06:
Can someone try with current next / @mluessi's recent optimization?

pug398, 2016-03-07:
Looks like optimization helped and cpu load at 75-78% running @1000hz. No more cpu warning but still occasional Event warning. That is with flight configuration including AH.

mlyle, 2016-03-07:
Yes, all of the targets have occasional event warnings.  So it works for basic flight-- that's great.

If it works for autotune we can close this issue, I think.

mlyle, 2016-03-21:
OK.  1KHz is not really recommended on Sparky1 but at least it seems to work.  Worth leaving the setting in IMO.

closed 2016-03-21 by mlyle