d-ronin/dRonin #783: GCS: Cleanup warnings on OSX

tracernz, 2016-03-11:
https://gist.github.com/tracernz/7049f259401b86d3d103

mlyle, 2018-02-25:
Can we consider this fixed with #1888 and #1890 (are the rest third party things we'll never fix?)

tracernz, 2018-02-25:
No, we still have plenty of our own 
```
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_date.cpp:174:22: warning: unused function 'qwtToJulianDay' [-Wunused-function]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_date.cpp:187:22: warning: unused function 'qwtFloorDiv64' [-Wunused-function]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_date.cpp:195:22: warning: unused function 'qwtFloorDiv' [-Wunused-function]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_date_scale_engine.cpp:30:27: warning: comparison of constant 8 with expression of type 'QwtDate::IntervalType' is always false [-Wtautological-constant-out-of-range-compare]
1 warning generated.
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_plot_tradingcurve.cpp:238:37: warning: comparison of constant 2 with expression of type 'QwtPlotTradingCurve::Direction' is always false [-Wtautological-constant-out-of-range-compare]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/libs/qwt/src/qwt_plot_tradingcurve.cpp:259:37: warning: comparison of constant 2 with expression of type 'QwtPlotTradingCurve::Direction' is always false [-Wtautological-constant-out-of-range-compare]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:189:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:229:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:533:57: warning: unused parameter 'result' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:533:71: warning: unused parameter 'sender' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:534:19: warning: unused parameter 'report_type' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:534:41: warning: unused parameter 'report_id' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:960:60: warning: unused parameter 'dev' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:960:69: warning: unused parameter 'string_index' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:960:92: warning: unused parameter 'string' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:960:107: warning: unused parameter 'maxlen' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:967:68: warning: unused parameter 'dev' [-Wunused-parameter]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/rawhid/hidapi/hidapi_mac.c:344:32: warning: unused function 'remove_from_list_by_path' [-Wunused-function]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:42:21: warning: 'shortName' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:43:21: warning: 'boardDescription' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:44:18: warning: 'queryCapabilities' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:45:21: warning: 'getBoardPicture' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:46:21: warning: 'getHwUAVO' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:47:18: warning: 'isInputConfigurationSupported' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:48:18: warning: 'setInputType' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:49:41: warning: 'getInputType' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:50:17: warning: 'queryMaxGyroRate' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/plugins/boards_dronin/seppuku.h:51:25: warning: 'getAdcNames' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
[... cut ...]
/Users/_buildbot/worker/mac_dronin/build/ground/gcs/src/app/main.cpp:218:48: warning: 'MV_10_8' is deprecated: Use QOperatingSystemVersion [-Wdeprecated-declarations]
1 warning generated.
/Users/_buildbot/worker/mac_dronin/build/make/firmware-common.mk:46: warning: overriding commands for target `/Users/_buildbot/worker/mac_dronin/build/build/fw_revolution/pios_gcsrcvr.o'
/Users/_buildbot/worker/mac_dronin/build/make/firmware-common.mk:46: warning: ignoring old commands for target `/Users/_buildbot/worker/mac_dronin/build/build/fw_revolution/pios_gcsrcvr.o'
```