d-ronin/dRonin #785: Altitude Hold not working on Naze32 running dronin

kjoy84, 2016-03-11:
Does anyone on here have experience using the "altitude hold" function on the Naze32 running Dronin. I have put a Naze32 10POF (I think rev 6) that has a magnetometer and barometer on my tricopter. I am using Dronin, not baseflight or cleanflight. The autotune function works great. My problem is that I can't get the altitude hold function to work. The altitude module enables but when add the altitude hold function to my flight mode switch the FC goes into a system configuration error state and will not arm. And it doesn't seem correct even if I am disarm in a different flight mode. Any ideas?

I have two Naze32's, both are doing the same thing.

dustin, 2016-03-11:
The baro isn't currently supported.

kjoy84, 2016-03-16:
Thank you both

closed 2016-03-16 by kjoy84