d-ronin/dRonin #786: GCS:Modules: Disable checkboxes for modules unavailable on current hardware

tracernz, 2016-03-11:
Should help with e.g. #785 where the module appears enabled but there's nobody listening to the module adminstate on the flight side.

jpbarraca, 2016-04-06:
On the current next this seems to be already implemented. 
The following image shows the module interface in a CC3D with some modules disabled.
Oddly, there is no GPS but there is geofence...?

[image]

mlyle, 2016-04-06:
The greyed ones are the ones that are either:

* turned on or off on a different GCS page
* automatically turn on when you configure a port to use them

tracernz, 2016-04-06:
I'll separate the auto-enabled ones in the UI soon.

mlyle, 2016-04-10:
I think my preferred approach for this would be--

* alter task status, to have 'not present', not running, or running status.
* have tasks default to 'not present'
* add a call to set a tasks status from not present -> not running
* have all modules call that at init time
* have GCS react to this appropriately.

tracernz, 2016-04-11:
Yes, I'd like to do that. Doesn't work for F1 though...

mlyle, 2016-06-12:
Fixed by #993

closed 2016-06-12 by mlyle