d-ronin/dRonin #805: Request: Ability to have external mag on Revo

rabno1, 2016-03-15:
Wouldn't even know where to start to get this working myself but would love to be able to use the external mag in my cheapo GPS with my revolution fc.  I believe the mag uses I2C and would be great if we could use the flexi port to connect it to.
Ill do anything i can to help.

jihlein, 2016-03-15:
I actually have this done, but not tested. Let me rebase and make sure it still builds, then I'll push a PR. Someone with a Revolution will have to test.

mlyle, 2016-03-16:
@rabno1-- do you have an external mag and are you able to confirm that https://github.com/d-ronin/dRonin/pull/806#issuecomment-197154629  works?

Note that the magnetometer heading shows up as a red stripe in the primary flight display on the flight data tab.

rabno1, 2016-03-16:
Wow that was quick. Ill need to wire up the mag but ill get on it and test it out. Ill report back ASAP.

PR #806 (2016-03-17): Flight: Add external magnetometer capability to Revolution target.
Adds the capability to use an external I2C magnetometer on the Revolution Flexiport when the Flexiport is configured as an I2C bus.

Needs testing by someone with Revolution hardware and external I2C magnetometer.

~~Addresses #805.~~ edit by mpl: fixes #805

---
This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/806)
closed 2016-03-17 by mlyle