d-ronin/dRonin #821: Known issues with upgrade meta-issue

mlyle, 2016-03-21:
After #801 lands, known items for cleanup.

* [ ] Blank flight controllers need to be handled better
* [ ] Completion screen should look better
* [ ] Issues with window height / focus, though these are problematic to address
* [ ] Verification of availability of cloud service, etc, in the beginning
* [ ] Better text on first dialog asking about upgrade.
* [ ] Need to not upgrade pipxtreme this way, or bug user in the same way about Naze32 or not over USB, etc...

closed 2016-03-30 by mluessi

tracernz, 2016-03-30:
Should it be a modal dialog? I see it sometimes ends up behind the main window (e.g. if an error occurs).

mlyle, 2016-03-30:
It is setmodal.  There are odd things with modalness and our main window.  (if you have more than one modal thing, Qt acts weird, etc).

PR #851 (2016-03-30): gcs: upgrade: improvements for blank/unknown FC revs
* handle blank flight controllers better
* test availability of cloud service
* warn user if hash not trivially upgradable

fixes #821 

---
This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/851)