d-ronin/dRonin #825: pipxtreme firmware broken in next?

rabno1, 2016-03-22:
Just been trying out the next build and it seems like the pipxtreme firmware is broken. Im trying to load it onto my oplink but it just goes into a boot loop once flashed. Ive tried my own built firmware and also the firmware from the build at the bottom of https://github.com/d-ronin/dRonin/pull/801
I can go back to the stable release and it works again ok. Ive tried using a stable version of the GCS just incase it was that and still get into a boot loop.

mlyle, 2016-03-22:
@pug398-- have you used current next with pipxtreme?  I'll be able to play with this in a couple days

pug398, 2016-03-22:
20160320 7fe1 is working. I was planning to upgrade sparky2/oplm to current next anyway so will let you know about newest in a little bit.

rabno1, 2016-03-22:
@pug398 where can i find that build and ill give it a try?

mlyle, 2016-03-23:
That's just a couple of days ago--- so very little has happened-- though I guess it's possible #787 has regressed it.

pug398, 2016-03-23:
Hmm. Apparently I am mistaken as it looks like I had 20160312 3f2c build on oplm and only upgraded sparky2. Double checking now but what I have on hand jumps from 3/12 build (working) to 3/16 build (bootloop).

mlyle, 2016-03-23:
OK, thanks for the confirmation that we're regressed.

rabno1, 2016-03-23:
ive got one from 3/16 that works fine with mine

pug398, 2016-03-23:
next 20160316-ea015996 or something you built yourself?

rabno1, 2016-03-23:
came from here https://github.com/d-ronin/dRonin/pull/806#issuecomment-197154629

mlyle, 2016-03-23:
Builds in PRs aren't really too helpful because I can't easily tell what they're built with.

rabno1, 2016-03-23:
Ah right! sorry this is all new to me, I thought those builds would be the same as the latest next build at the time but with the changes in the PR added.

mlyle, 2016-03-23:
@rabno1, they are, but they're usually hashes not in my repo and things I can't use for a range check.

I'm sorry if I'm a bit surly-- I do appreciate the report and feedback-- just having a lot of trouble with other things now.

PR #827 (2016-03-23): radiocombridge: remove doubled watchdog reg
@rabno1, @pug398-- does this fix things?  if it does, it's a regression we picked up from #807 

hoped: fixes #825

---
This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/827)
closed 2016-03-23 by dustin